alexpevzner/sane-airscan #268: EPSON ET-4750: confirmed to work

daniel-sch, 2023-01-01:
```
% scanimage -L
device `airscan:w0:EPSON ET-4750 Series' is a WSD EPSON ET-4750 Series ip=x.x.x.x
```

Works like a charm :-)

alexpevzner, 2024-03-07:
This is WSD. WSD works only via network (Ethernet or WiFi).

jobitobi, 2024-03-08:
Hallo Alex, habe dein Profil gestalkt und gesehen du wohnst/kommst aus Berlin. Deshalbe versuche ich es auf Deutsch.  Ich bringe inzwischen den Drucker dazu den WSD Rechner zu erkennen. Aber wenn ich scanne weiß ich nicht wo die Scans abgelegt werden. Brauche ich auf dem Host (WSD-Rechner) noch ein Programm=

alexpevzner, 2024-03-09:
Hi @jobitobi,

I'm from Moscow, Russia. I don't know German, but, thanks to Google Translate, I can read it,

Regarding your question, there are, basically, two methods to use scanner:
1. Pull-scan. Scanning initiated by some program at the computer and that program manages received images. Typically, these programs have GUI.
2. Push-scan. Scanning initiated by some manipulations with scanner, resulting image is sent to computer and saved there.

For Linux it is mostly typical to use pull-scan (for Windows, as far as I know, both methods are in use). So you need some program to scan with `sane-airscan` (airscan is just a driver). There are many programs to choose from. For example `simple-scan`, `xsane`. Even Libre Office has built-in scanning support and can scan and insert inages directly into the document being edited.

P.S. I feel now out conversation looks more like discussion that bug report, so I reopen it and convert into discussion