alexpevzner/sane-airscan #269: Samsung C640w hangs when scannin in gray mode

letorbi, 2023-01-17:
Hello,

I have a Samsung C640w, which works perfectly well, when I scan images in color mode. However, the scanning process "hangs", when I try to scan in gray mode.

From the sound of the scanner I would say, the scanning starts, but then stops after one or two seconds and the printer becomes active for another second. After that nothing more happens. I've waited for about two hours until I cancelled the command (you will notice the `^C` in the *gray.out* file). 

I've tried *simple-scan* as well as *scanimage* and both behave the same. For the attached tests I've used the following commands:

* `scanimage --device "airscan:w1:Samsung C460 Series (Hermes)" --format=png --output-file test.png --progress --mode Color`
* `scanimage --device "airscan:w1:Samsung C460 Series (Hermes)" --format=png --output-file test.png --progress --mode Gray`

Here are the logs and traces for scanning a color and a gray image:

* [color.zip](https://github.com/alexpevzner/sane-airscan/files/10439286/color.zip)
* [gray.zip](https://github.com/alexpevzner/sane-airscan/files/10439288/gray.zip)

Each archive contains the airscan-trace as well as the console output in a *\*.out* file.

Please write, if you need more info. I am willing to help :)

alexpevzner, 2023-01-18:
Hi @letorbi,

in the `sane-airscan` log I see that scanning was cancelled nearly immediately after start, in the Grayscale mode.

Looks like some `scanimage` problem. Could you please retest with some another client? Say, `simple-scan` or `xsane`

letorbi, 2023-01-18:
Sure, here are the logs and traces for *simple-scan*, *xsane* and again for *scanimage*.

For what it's worth: I've recognized that scanning in gray-mode works, if I set the resolution to 200 dpi or less. I've also attached the traces and logs for a scan with *scanimage* at 200 dpi.

* [scanimage-color.zip](https://github.com/alexpevzner/sane-airscan/files/10451219/scanimage-color.zip)
* [scanimage-gray.zip](https://github.com/alexpevzner/sane-airscan/files/10451220/scanimage-gray.zip)
* [scanimage-gray200.zip](https://github.com/alexpevzner/sane-airscan/files/10451221/scanimage-gray200.zip)
* [simple-scan-color.zip](https://github.com/alexpevzner/sane-airscan/files/10451222/simple-scan-color.zip)
* [simple-scan-gray.zip](https://github.com/alexpevzner/sane-airscan/files/10451225/simple-scan-gray.zip)
* [xsane-color.zip](https://github.com/alexpevzner/sane-airscan/files/10451226/xsane-color.zip)
* [xsane-gray.zip](https://github.com/alexpevzner/sane-airscan/files/10451227/xsane-gray.zip)

letorbi, 2023-04-25:
FYI After a completely new installation of the whole system, the problem disappeared, so I'm closing this issue.

closed 2023-04-25 by letorbi