alexpevzner/sane-airscan #281: Pantum M7100DN series: Error during device I/O

matwey, 2023-04-05:
Hello,

I am running `sane-airscan-unstable-1620822232`, `skanlite-21.12.3`, and have the following issues with Pantum M7100DN series when trying to scan.

I can start scanning process and then the scanner is running and making noise. When the process ends I see only "Error during device I/O" from skanlite and nothing more.

Trace files are attached here:
[skanlite-Pantum-M7100DN-series.log](https://github.com/alexpevzner/sane-airscan/files/11157891/skanlite-Pantum-M7100DN-series.log)
[skanlite-Pantum-M7100DN-series.tar.gz](https://github.com/alexpevzner/sane-airscan/files/11157897/skanlite-Pantum-M7100DN-series.tar.gz)

alexpevzner, 2023-04-05:
Hi @matwey,

the problem occurs because your device claims that it has sent image in the PNG format, while actually format is JPEG. Obviously, PNG decoder returns a error in attempt to unpack it.

Looks like I can't trust a device and have to implement image format autodetect

BTW, why do you use -unstable branch?

matwey, 2023-04-05:
I've tried stable first, it didn't work also. Then I've tried unstable to see if the issue already resolved.

alexpevzner, 2023-04-05:
It's the first device I've ever seen that reports an image format different from what it actually sends.

(it correctly reports MIME type, but MIME type is not reliable on other devices, so I don't use it. But image format types reported in the list of supported types and in response to the scan request were reliable so far)

matwey, 2023-04-05:
Maybe it is just a bug in the firmware, but I don't see any updates for this device firmware.

alexpevzner, 2023-04-05:
It's definitely bug in firmware. Not the first one, not the last :-)

For this but it is at least a quite clear how to implement a workaround

alexpevzner, 2023-07-06:
Hi @matwey,

I've hopefully fixed this problem by adding dynamic detection of the actual image format supported, basedo on a image data signature

I would appreciate if you can build `sane-airscan` by yourself and test it with your device to confirm that fix works for you

matwey, 2023-07-07:
Thank you. Probably, I will be able to test it in the next Monday.

alexpevzner, 2023-07-07:
OK. Building `sane-airscan` is a really simple task. And I suggest you not to do `make install`, but instead figure out the location where `libsane-airscan.so.1` is installed on your system (usually it is `/usr/lib64/sane/libsane-airscan.so.1` or `/usr/lib/sane/libsane-airscan.so.1`, depending on your distro) and replace it with the symbolic link to the `libsane-airscan.so.1` you've build by yourself

matwey, 2023-07-10:
Scanning works now. Thank you.

alexpevzner, 2024-02-04:
Fixed, closed

closed 2024-02-04 by alexpevzner