alexpevzner/sane-airscan #284: Epson ET-2710 detected as both Epson ET-2710 (airscan) and Epson L3150 (epsonds)

Dremor, 2023-04-22:
Hi.

I'm currently building this project in order to propose it as a new package for openSUSE. I compiled the latest release as well as the latest master commit

In both case, my printer, an Epson ET-2710, shows as two differently named scanner.

```
$ scanimage -L
device `epsonds:net:192.168.0.232' is a Epson L3150 Series ESC/I-2
device `airscan:w1:EPSON ET-2710 Series' is a WSD EPSON ET-2710 Series ip=192.168.0.232
``` 

Both reference seem to be the same printer ([source](https://download.epson-europe.com/pub/download/6306/epson630651eu.pdf)), but for different market and protocols.

This may be happening because I have configured the web print thing, but that was long ago.

Am I right ?

Dremor, 2023-04-22:
Adding airscan-discover output :
```
$ airscan-discover
[devices]
  EPSON ET-2710 Series = http://192.168.0.232:80/WDP/SCAN, WSD
  EPSON ET-2710 Series = http://[fe80::xxxxxxxxxxxxxxxxxxxxx]:80/WDP/SCAN, WSD
```

alexpevzner, 2023-07-06:
Hi, @Dremor,

it's not a bug. Two drivers independently discovered your device, `sane-airscan` and `sane-epsonsd`, because your printer speaks both protocols.

Unfortunately, in the SANE architecture drivers don't communicate to each other and there is no central agent that can combine discovery output from several drivers and filter out duplicates.

closed 2023-07-06 by alexpevzner