alexpevzner/sane-airscan #285: Ricoh P311 doesn't work with native driver or AirScan

natanschneider, 2023-04-25:
I have a Ricoh P311, i was trying to get it working with the native driver, wasn't going nowhere. Than i tried with sane-airscan, sane connects with the printer, turns it on, but instead of scanning, it starts to print a bunch of random symbols.
Heres the output:
```
> scanimage --device "airscan:e0:Ricoh P311" --mode Color --format=png --output-file teste.png --progress

API: sane_init(): called
loading configuration file /etc/sane.d/airscan.conf
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 172.26.51.164
WSDD:  + fe80::215:5dff:fe0c:a653%2
WSDD: 172.26.51.164: started discovery, UDP port=41842
WSDD: fe80::215:5dff:fe0c:a653: started discovery, UDP port=45768
API: sane_open("e0:Ricoh P311"): called
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: device_list wait: OK
Ricoh P311: device created
Ricoh P311: DEVICE_STM_OPENED->DEVICE_STM_PROBING
Ricoh P311: using protocol "eSCL"
Ricoh P311: HTTP GET https://192.168.31.49:9100/eSCL/ScannerCapabilities
Ricoh P311: HTTP using timeout: 5000 ms
Ricoh P311: HTTP resolving 192.168.31.49 9100
Ricoh P311: HTTP trying 192.168.31.49:9100
Ricoh P311: HTTP fdpoll: {--} -> {-w}
Ricoh P311: HTTP fdpoll: {-w} -> {r-}
WSDD: fe80::215:5dff:fe0c:a653: done discovery
WSDD: 172.26.51.164: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
Ricoh P311: HTTP 192.168.31.49:9100: gnutls_handshake(): An unexpected TLS packet was received.
Ricoh P311: HTTP GET https://192.168.31.49:9100/eSCL/ScannerCapabilities: An unexpected TLS packet was received.
Ricoh P311: scanner capabilities query: An unexpected TLS packet was received.
Ricoh P311: DEVICE_STM_PROBING->DEVICE_STM_PROBING_FAILED
Ricoh P311: removed from device table
Ricoh P311: closed protocol "eSCL"
Ricoh P311: device destroyed
API: sane_open("e0:Ricoh P311"): Error during device I/O
scanimage: open of device airscan:e0:Ricoh P311 failed: Error during device I/O
API: sane_exit(): called
API: sane_exit(): OK
```

alexpevzner, 2023-07-06:
Hi @natanschneider,

what `airscan-discover` prints?

alexpevzner, 2023-07-07:
Ping. Is this issue still relevant?

alexpevzner, 2024-02-04:
No response...

closed 2024-02-04 by alexpevzner