alexpevzner/sane-airscan #289: Does not try eSCL if WS fails

mikhailnov, 2023-05-18:
Hello!
I have tested sane-airscan with Kyocera ECOSYS M2640idw, which works via eSCL only by default.
When I had the `wsdd` (https://github.com/christgau/wsdd) daemon running, I got a message:
`WSDD: bind(0.0.0.0:3702): Address is already in use`
It is because wsdd listens 0.0.0.0:3702.
No scanners were found.
Then I added `ws-discovery = off` into `/etc/sane.d/airscan.conf` and eSCL Kyocera was found.
Seems that sane-airscan does not try eSCL if WS discovery fails, can it continue and try eSCL?

alexpevzner, 2023-05-18:
Hi, @mikhailnov,

A while ago the similar issue was already reported to the WSDD project: https://github.com/christgau/wsdd/issues/95

I was participating in the discussion and in the investigation

We came to agreement that issue is at their side, and issue seems to be fixed there. Strange, that it came back again.

The patch that you've proposed will lead to instability in the `sane-airscan` work. Sometimes it will work with WSD properly, sometimes it will not work. And it will be hard to guess for the end user what is going on.

In a case of conflict or if WS-Discovery is just needed, `sane-airscan` allows to disable WS-Discovery via its configuration file (`/etc/sane.d/airscan.conf`):

```
[options]
ws-discovery = off
```

mikhailnov, 2023-05-19:
Thanks! I had wsdd 0.6.4, it does not have that fix. Great if hackery in sane-airscan is not required.

closed 2023-05-19 by mikhailnov