alexpevzner/sane-airscan #295: Brother MFC-2750DW isn't scanning duplex.

heinrichat, 2023-08-01:
Hi!

Bought completely new. Everything works flawlessly. 2750DW has duplex scanning capability. Using on Debian 10.

```
$ scanimage -V
scanimage (sane-backends) 1.0.31-debian; backend version 1.0.31
```

```
$ airscan-discover
[devices]
  Brother MFC-L2750DW series = http://192.168.1.92:80/eSCL/, eSCL
  Brother MFC-L2750DW series = http://192.168.1.92:80/WebServices/ScannerService, WSD
  Brother MFC-L2750DW series = http://[FE80::12B1:DFFF:FED3:45F7%252]:80/WebServices/ScannerService, WSD
  Brother MFC-L2750DW series = http://[FE80::12B1:DFFF:FED3:45F7%253]:80/WebServices/ScannerService, WSD
```

```
$ scanimage -A
scanimage: rounded value of br-x from 215.9 to 215.9
scanimage: rounded value of br-y from 296.926 to 296.926

All options specific to device `escl:http://192.168.1.92:80':
  Scan mode:
    --mode Lineart|Gray|Color [Color]
        Selects the scan mode (e.g., lineart, monochrome, or color).
    --resolution 100|200|300|600dpi [100]
        Sets the resolution of the scanned image.
    --preview[=(yes|no)] [no]
        Request a preview-quality scan.
    --preview-in-gray[=(yes|no)] [no]
        Request that all previews are done in monochrome mode.  On a
        three-pass scanner this cuts down the number of passes to one and on a
        one-pass scanner, it reduces the memory requirements and scan-time of
        the preview.
  Geometry:
    -l 0..210.481mm [0]
        Top-left x position of scan area.
    -t 0..291.507mm [0]
        Top-left y position of scan area.
    -x 5.41866..215.9mm [215.9]
        Width of scan-area.
    -y 5.41866..296.926mm [296.926]
        Height of scan-area.
    --source Flatbed|ADF|ADF Duplex [Flatbed]
        Selects the scan source (such as a document-feeder).
```

After placing two sheets of paper into the DADF... device tells me "ADF ready" and I issue following command...

```
$ scanimage -b --source "ADF Duplex"
scanimage: rounded value of br-x from 215.9 to 215.9
scanimage: rounded value of br-y from 355.6 to 355.6
Scanning infinity pages, incrementing by 1, numbering from 1
Scanning page 1
Scanned page 1. (scanner status = 5)
Scanning page 2
scanimage: sane_start: Invalid argument
Batch terminated, 1 page scanned
```

It creates only one file containing the "frontside" of the first page.

[Here](https://pastebin.com/raw/G0ZxtLet) is the debug output. I had to use pastebin, because github comment can have max 65536 chars... M$ apperantly is running low on disk space and has to skimp on storage.

And [here](https://pastebin.com/raw/LhHqqdaW) is the debug log.

Thank you for your help.

heinrichat, 2023-08-01:
Maybe duplicate of #225 even though I don't even get the odd numbers. Only one file and then exit.

heinrichat, 2023-08-01:
When I place two sheets of paper into the ADF and issue this command `\scanimage -b --batch-prompt --source="ADF Duplex" (the backslash is for ignoring an bash alias) the result is as follows:

`$ \scanimage -b --batch-prompt --source "ADF Duplex"
Output format is not set, using pnm as a default.
scanimage: rounded value of br-x from 215.9 to 215.9
scanimage: rounded value of br-y from 355.6 to 355.6
Scanning infinity pages, incrementing by 1, numbering from 1
Place document no. 1 on the scanner.
Press <RETURN> to continue.
Press Ctrl + D to terminate.

Scanning page 1
Scanned page 1. (scanner status = 5)
Place document no. 2 on the scanner.
Press <RETURN> to continue.
Press Ctrl + D to terminate.

Scanning page 2
scanimage: sane_start: Invalid argument
Batch terminated, 1 page scanned`

I press once return. The ADF starts scanning all the paper until the ADF is empty. Then it prompts me to again press return to scan the next page, but obviously the ADF is already empty. Duh!

Maybe this can help.

heinrichat, 2023-08-01:
I've already done that.

> And [here](https://pastebin.com/raw/G0ZxtLet) is the debug trace log.

alexpevzner, 2023-08-01:
Looks like you have both `sane-airscan` and `sane-escl` installed, and actually scan with `sane-escl`.

It may happen because you don't explicitly specify the device when invoking the `scanimage` command and the first device in the list appears to be from the `sane-escl` driver.

I suggest you to try to comment out the `escl` keyword in the `/etc/sane.d/dll.conf` file and retest.

heinrichat, 2023-08-01:
Dude... it works!

Finally... after Canon, Ricoh, Kyocera I found Brother which just works. Thank you!!

alexpevzner, 2023-08-01:
Glad to hear! :)

closed 2023-08-01 by alexpevzner