alexpevzner/sane-airscan #299: file airscan-devops.c: line 129 (devopt_choose_colormode): assertion failed: (wanted < NUM_ID_COLORMODE)

b10126, 2023-08-26:
Hi,
tried to get along with my HP Officejet 6600, which is eSCL compatible but it ends with
_file airscan-devops.c: line 129 (devopt_choose_colormode): assertion failed: (wanted < NUM_ID_COLORMODE)_

attached there is the log file from the trace.
anyone suggestions what the problem is?

thanks a lot,
benedikt

root@sane:~/airscan/trace# cat scanimage-HP-Officejet-6600.log
00:00:02.526: HP Officejet 6600: device created
00:00:02.526: HP Officejet 6600: DEVICE_STM_OPENED->DEVICE_STM_PROBING
00:00:02.526: HP Officejet 6600: using protocol "eSCL"
00:00:02.526: HP Officejet 6600: HTTP GET http://192.168.0.5/eSCL/ScannerCapabilities
00:00:02.526: HP Officejet 6600: HTTP using timeout: 5000 ms
00:00:02.526: HP Officejet 6600: HTTP resolving 192.168.0.5 80
00:00:02.526: HP Officejet 6600: HTTP trying 192.168.0.5:80
00:00:02.534: HP Officejet 6600: HTTP 80 bytes sent
00:00:02.534: HP Officejet 6600: HTTP done request sending
00:00:02.583: HP Officejet 6600: HTTP 1753 bytes received
00:00:02.584: HP Officejet 6600: HTTP GET http://192.168.0.5/eSCL/ScannerCapabilities: got response headers (200)
00:00:02.584: HP Officejet 6600: HTTP 1448 bytes received
00:00:02.587: HP Officejet 6600: HTTP 1448 bytes received
00:00:02.591: HP Officejet 6600: HTTP 2896 bytes received
00:00:02.596: HP Officejet 6600: HTTP 193 bytes received
00:00:02.596: HP Officejet 6600: HTTP done response reception
00:00:02.596: HP Officejet 6600: HTTP GET http://192.168.0.5/eSCL/ScannerCapabilities: 200 OK
==============================
GET http://192.168.0.5/eSCL/ScannerCapabilities
Connection: close
Host: 192.168.0.5

Status: 200 OK
Server: HP HTTP Server; HP Officejet 6600 - CZ155A; Serial Number: CN47Q8R0XH05RN; Margarita_mid_pendiag_mp2 Built:Thu Mar 12, 2020 06:18:50AM {MIM5CN2011AR, ASIC id 0x003a0008}
Content-Type: text/xml
Content-Length: 7433
Cache-Control: must-revalidate, max-age=0
Pragma: no-cache

<comment/>
<scan:ScannerCapabilities xmlns:scan="http://schemas.hp.com/imaging/escl/2011/02/08" xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://schemas.hp.com/imaging/escl/2011/02/08 eSCL-0_98.xsd">
  <pwg:Version>1.0</pwg:Version>
  <pwg:MakeAndModel>Officejet 6600 e-All-in-One</pwg:MakeAndModel>
  <pwg:SerialNumber>CN47Q8R0XH05RN</pwg:SerialNumber>
  <scan:ColorProfiles>
    <scan:ColorProfile>
      <scan:ColorReference>Grayscale8</scan:ColorReference>
      <scan:Color>Grayscale</scan:Color>
      <scan:BitsPerColorChannel>8</scan:BitsPerColorChannel>
      <scan:Formats>
        <pwg:DocumentFormat>application/octet-stream</pwg:DocumentFormat>
        <pwg:DocumentFormat>image/jpeg</pwg:DocumentFormat>
      </scan:Formats>
      <scan:ChromaticSpaces>
        <scan:GrayscaleSpaces>
          <scan:GrayscaleSpace>NTSC</scan:GrayscaleSpace>
          <scan:GrayscaleSpace>GrayCcdEmulated</scan:GrayscaleSpace>
        </scan:GrayscaleSpaces>
      </scan:ChromaticSpaces>
      <scan:ImageTransforms>
        <scan:ImageTransform>Sharpen</scan:ImageTransform>
        <scan:ImageTransform>NoiseRemoval</scan:ImageTransform>
      </scan:ImageTransforms>
    </scan:ColorProfile>
    <scan:ColorProfile>
      <scan:ColorReference>RGB24</scan:ColorReference>
      <scan:Color>Color</scan:Color>
      <scan:BitsPerColorChannel>8</scan:BitsPerColorChannel>
      <scan:Formats>
[... 154 lines cut ...]
    <scan:Step>1</scan:Step>
  </scan:SharpenSupport>
  <scan:ThresholdSupport>
    <scan:Normal>128</scan:Normal>
    <scan:Step>1</scan:Step>
  </scan:ThresholdSupport>
</scan:ScannerCapabilities>

00:00:02.597: HP Officejet 6600: HTTP GET http://192.168.0.5/eSCL/ScannerCapabilities: OK
00:00:02.597: HP Officejet 6600: ===== device capabilities =====
00:00:02.597: HP Officejet 6600:   Size units:       300 DPI
00:00:02.597: HP Officejet 6600:   Protocol:         eSCL
00:00:02.597: HP Officejet 6600:   Sources:          Flatbed, ADF

00:00:02.597: HP Officejet 6600:   Flatbed:
00:00:02.597: HP Officejet 6600:     Min window:  8x8 px, 0.68x0.68 mm
00:00:02.597: HP Officejet 6600:     Max window:  2550x3508 px, 215.90x297.01 mm
00:00:02.597: HP Officejet 6600:     Color modes:
00:00:02.597: HP Officejet 6600:     Formats:

00:00:02.598: HP Officejet 6600:   ADF:
00:00:02.598: HP Officejet 6600:     Min window:  8x8 px, 0.68x0.68 mm
00:00:02.598: HP Officejet 6600:     Max window:  2550x4200 px, 215.90x355.60 mm
00:00:02.598: HP Officejet 6600:     Color modes:
00:00:02.598: HP Officejet 6600:     Formats:

alexpevzner, 2023-08-26:
Hi @b10126,

looks like you are trying to modify `sane-airscan` code to work with your printer, correct?

Unmodified driver will not allow empty list of color modes due to check at `airscan-escl.c:326`.

Let's discuss your modifications, please.

b10126, 2023-09-02:
hi @alexpevzner,

sorry for my late reply, started this thing on weekends and work got me fully under control....
i'm far away from modifying, more or less testing if it would work, because some other HP Officejet Models are in your list. but reality hit me ;-)

so, but we've one result: HP Officejet 6600 doesn't work.

stupid question: what needs to be done to get it work?

debiantriage, 2023-09-02:
@b10126 Prove outpurs for

`avahi-browse -rt _ipp._tcp`
`avahi-browse -rt _uscan._tcp`
`scanimage -L`

b10126, 2023-09-02:
@debiantriage  here we go....  

`avahi-browse -rt _ipp._tcp`

> root@sane:/usr/bin# avahi-browse -rt _ipp._tcp
> + (null) IPv4 Officejet 6600 [D2CB2F]                       Internet Printer     local
> = (null) IPv4 Officejet 6600 [D2CB2F]                       Internet Printer     local
>    hostname = [HP288023D2CB2F.local]
>    address = [192.168.0.5]
>    port = [631]
>    txt = ["Scan=T" "Duplex=F" "Color=T" "UUID=1c852a4d-b800-1f08-abcd-288023d2cb2f" "note=" "adminurl=http://HP288023D2CB2F.local." "mac=28:80:23:d2:cb:2f" "priority=20" "usb_MDL=Officejet 6600" "usb_MFG=HP" "product=(HP Officejet 6600)" "ty=Officejet 6600" "URF=CP1,MT1-2-8-9-10-11,OB9,OFU0,PQ3-4-5,RS300-600,SRGB24,W8,DEVW8,DEVRGB24-48,ADOBERGB24-48,IS1" "rp=ipp/printer" "pdl=application/vnd.hp-PCL,image/jpeg,application/PCLm,image/urf" "qtotal=
1" "txtvers=1"]

`avahi-browse -rt  _uscan._tcp`
> no output

`scanimage -L`

> root@sane:/usr/bin# scanimage -L
> API: sane_init(): called
> loading configuration file /etc/sane.d/airscan.conf
> /etc/sane.d/airscan.conf:16: device already defined
> /etc/sane.d/airscan.conf:17: device already defined
> /etc/sane.d/airscan.conf:18: device already defined
> zeroconf:   HP Officejet 6600 = http://192.168.0.5/eSCL, eSCL
> MDNS: AVAHI_CLIENT_S_RUNNING
> WSDD: netif addresses update:
> WSDD:  + 192.168.0.71
> WSDD: 192.168.0.71: started discovery, UDP port=58229
> MDNS: browse/ipv4 "Officejet 6600 [D2CB2F]": AVAHI_BROWSER_NEW _ipp._tcp
> API: sane_get_devices(): called
> MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipp._tcp
> MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipps._tcp
> MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscan._tcp
> MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscans._tcp
> zeroconf: zeroconf_device_list_get: requested
> zeroconf: device_list wait: requested
> zeroconf: device_list wait: DNS-SD not finished...
> MDNS: browse/ipv4 "Officejet 6600 [D2CB2F]": AVAHI_BROWSER_NEW _scanner._tcp
> MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _scanner._tcp
> MDNS: resolve/ipv4 "Officejet 6600 [D2CB2F]": AVAHI_RESOLVER_FOUND _ipp._tcp
> WSDD: directed probe: trying if=?, addr=192.168.0.5
> WSDD: HTTP POST http://192.168.0.5/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
> WSDD: HTTP resolving 192.168.0.5 80
> WSDD: HTTP trying 192.168.0.5:80
> MDNS: resolve/ipv4 "Officejet 6600 [D2CB2F]": AVAHI_RESOLVER_FOUND _scanner._tcp
> zeroconf: found urn:uuid:1c852a4d-b800-1f08-abcd-288023d2cb2f
> zeroconf:   method:    ZEROCONF_MDNS_HINT
> zeroconf:   interface: 16 (?)
> zeroconf:   name:      Officejet 6600 [D2CB2F]
> zeroconf:   model:     Officejet 6600
> zeroconf:   addresses:
> zeroconf:     192.168.0.5
> zeroconf:   device:    0000 (created)
> WSDD: directed probe: trying if=?, addr=192.168.0.5
> WSDD: directed probe: already in progress, skipping
> WSDD: HTTP 757 bytes sent
> WSDD: HTTP done request sending
> zeroconf: device_list wait: DNS-SD not finished...
> zeroconf: found urn:uuid:1c852a4d-b800-1f08-abcd-288023d2cb2f
> zeroconf:   method:    ZEROCONF_WSD
> zeroconf:   interface: 127 (eth0)
> zeroconf:   name:      -
> zeroconf:   model:     -
> zeroconf:   addresses:
> zeroconf:   protocol:  WSD
> zeroconf:   endpoints:
> zeroconf:   device:    0001 (created)
> zeroconf: device_list wait: DNS-SD not finished...
> WSDD: HTTP 317 bytes received
> WSDD: HTTP POST http://192.168.0.5/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: got response headers (200)
> WSDD: HTTP 1448 bytes received
> WSDD: HTTP 695 bytes received
> WSDD: HTTP done response reception
> WSDD: HTTP POST http://192.168.0.5/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: 200 OK
> WSDD: HTTP POST http://192.168.0.5/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: OK
> WSDD: directed probe: resolver not found for interface 16
> MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipp._tcp
> MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipps._tcp
> MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscan._tcp
> zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
> MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscans._tcp
> zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
> zeroconf: device_list wait: DNS-SD not finished...
> MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _scanner._tcp
> zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
> zeroconf: device_list wait: waiting for any proto for 'Officejet 6600 [D2CB2F]' (0)
> WSDD: 192.168.0.71: done discovery
> zeroconf: ZEROCONF_WSD: initial scan finished
> zeroconf: device_list wait: OK
> zeroconf: zeroconf_device_list_get: building list of devices
> zeroconf: Officejet 6600 [D2CB2F] (0): can: none, use: none
> zeroconf: Officejet 6600 [D2CB2F] (0): skipping, none of supported protocols discovered
> zeroconf: urn:uuid:1c852a4d-b800-1f08-abcd-288023d2cb2f (1): can: none, use: none
> zeroconf: urn:uuid:1c852a4d-b800-1f08-abcd-288023d2cb2f (1): skipping, none of supported protocols discovered
> zeroconf: zeroconf_device_list_get: resulting list:
> zeroconf:   eSCL  "e0:HP Officejet 6600"
> API: sane_get_devices(): done
> device `airscan:e0:HP Officejet 6600' is a eSCL HP Officejet 6600 ip=192.168.0.5
> API: sane_exit(): called
> zeroconf: device gone urn:uuid:1c852a4d-b800-1f08-abcd-288023d2cb2f
> zeroconf:   method:    ZEROCONF_WSD
> zeroconf:   interface: 127 (eth0)
> zeroconf: device gone urn:uuid:1c852a4d-b800-1f08-abcd-288023d2cb2f
> zeroconf:   method:    ZEROCONF_MDNS_HINT
> zeroconf:   interface: 16 (?)
> API: sane_exit(): OK

debiantriage, 2023-09-03:
Amazon have this device as available on December 14, 2011. This is about two years before Apple's AirScan appeared. It is not surprising there isn't any uscan advertised.

The manual indicates a Webscan feature is offered. I would guess this is why eSCL is on the machine. Th e protocol implementation  may be good enough for Webscan but not for other purposes?

Is scanning possible with `simple-scan "airscan:e0:HP Officejet 6600"` or `xsane "airscan:e0:HP Officejet 6600"`?

alexpevzner, 2023-09-04:
Dear Gentlemen,

this device speaks eSCL version 1.0, which is quite different protocol that eSCL 2.x, supported by airscan (and published by Mopria).

I was wondering why airscan asserts instead of rejecting the device with an error. Looks like it is quite old version of airscan, I've already fixed this assertion (but it is not the same as supporting eSCL 1.0).

In theory, eSCL 1.0 could be supported in the airscan. But somebody else, not me, should contribute this implementation. I simply don't have any eSCL 1.0 device around me to work with.

It is not obvious though how practical this implementation could be. As far as I know, only few old-fashion HP devices support this version of the protocol. All modern devices support either eSCL 2.x or WSD or even both.

b10126, 2023-09-04:
Hi everyone, thanks for this really cool support. I don't think it is worth to spend so much time in running after an old protocol, which is not longer used and depreciated. Therefore I'll stick to my old solution (good old hp scan software) or will invest in new equipment which is much more worth than spending time on old fashioned connections.
Have a good day,
Benedikt

closed 2023-09-04 by b10126