alexpevzner/sane-airscan #300: HP Scanjet 200. Does not work

aleksey1831981, 2023-09-20:
Александр, приветствую.

Пишу на русском, вроде как на этом языке тут могут понять)
Когда-то решали тут с вами вопрос по работе офисного сканера. Проблему решили.
Теперь вот купил дочке планшетный сканер HP Scanjet 200 для учебы. Но этот сканер с Linux не захотел работать.

```
$ scanimage -L

No scanners were identified. If you were expecting something different,
check that the scanner is plugged in, turned on and detected by the
sane-find-scanner tool (if appropriate). Please read the documentation
which came with this software (README, FAQ, manpages).
```

```
$ sudo ipp-usb check

No IPP over USB devices found
```
Данные о системе:
```
System:
  Kernel: 5.15.0-84-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 5.4.12
    tk: GTK 3.24.33 wm: Mutter dm: LightDM Distro: Linux Mint 21 Vanessa base: Ubuntu 22.04 jammy
```

Возможно ли как то заставить работать этот сканер в UNIX-подобных операционных системах?

alexpevzner, 2023-09-20:
Здравствуйте, Алексей,

я не знаю, как помочь Вам с этим сканером. Судя по всему, он не поддерживает IPP over USB, поэтому он не будет работать ни с `ipp-usb`, ни с `sane-airscan`.

Кроме того, я бегло поискал по интернету. кажется, у многих пользователей существует проблема с поддержкой этого сканера в Linux

Если у вас есть еще такая возможность, могу посоветовать вернуть этот сканер в магазин и вместо него приобрести Canon LiDE 300: https://www.dns-shop.ru/product/54edc35a34c4ed20/skaner-canon-canoscan-lide-300/

Цена у него какая-то более-менее сравнимая, параметры тоже, и про него известно, что он поддерживает IPP over USB и eSCL, т.е., работает с моими драйверами.

closed 2023-09-20 by alexpevzner

aleksey1831981, 2023-09-20:
Александр, спасибо за ответ и за совет.
Буду думать, что теперь делать с этим сканером.