alexpevzner/sane-airscan #304: Lexmark MX310dn and sane-airscan does not work :)

aleksey1831981, 2023-11-02:
Александр, приветствую.
Итак я приобрел новый девайс: Lexmark MX310dn, взамен залоченного под винду сканера.
MX310dn сканер не определялся никак без установки драйверов. А после установки драйверов с сайта Lexmark, сканер стал виден в Simple Scan и Scan2PDF.  

Simple Scan определяет сканер как "Lexmark Network". При попытке начать сканирование, указывает, что указан неверный IP адрес. 

Scan2PDF дает возможность указать IP адрес сканера, но при попытке сканирования запускает фирменную утилиту Lexmark, которая весьма убога по функционалу.

Привожу вывод команды ``airscan-discover``
```
$ airscan-discover
[devices]
``` 
Это всё)

``scanimage -L``
```
$ scanimage -L
device `lexmark_nscan:libnet/SPECIFY_DEVICE' is a Lexmark Network Scanner
``` 

Так же я прикрепляю логи ~/airscan/trace
[simple-scan-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/13244013/simple-scan-zeroconf.log)

Буду признателен за помощь.

aleksey1831981, 2023-11-02:
И, да, обновил прошивку МФУ до последней актуальной.

alexpevzner, 2023-11-02:
Добрый вечер!

Я совершенно не уверен, что этот принтер поддерживает WSD/eSCL, но меня смущают некоторые вещи в логе. Я бы порекомендовал проверить в настройках принтера (в веб-консоли):
* Network/Ports > Standard Network > Standard Network Setup > TCP/IP
  * Set Hostname - какое-то осмысленное имя (возможно, оно уже есть по умолчанию)
  * Enable mDNS - Yes
 
Очень странно, что Avahi (mDNS) его никак не видит. Как минимум, он долже был бы виден как принтер...

aleksey1831981, 2023-11-02:
Да, у меня действительно было отключено mDNS. Я включил это. Перезагрузил устройство. В Simple Scan появился второй сканер: Lexmark Ethernet. Сканирование по прежнему не получается. 
Если выбрать сканер Lexmark Network - ошибка, что не указан IP адрес.
Если выбрать Lexmark Ethernet, открывается опять эта убогая фирменная программа от Lexmark, в которой можно сканировать по одной странице.

Ниже вывод `scanimage -L`
И прикрепил новые логи сканирования в Lexmark Network и Lexmark Ethernet
```
$ scanimage -L
API: sane_init(): called
loading configuration file /etc/sane.d/airscan.conf
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 2a00:62c0:2be8:8400::4cb
WSDD:  + 2a00:62c0:2be8:8400:aaf8:2303:5cb0:8dc
WSDD:  + 2a00:62c0:2be8:8400:b809:d6e1:3fad:8460
WSDD:  + fd13:4514:2d3b:0:865b:3b22:f116:7e3c
WSDD:  + fd13:4514:2d3b:0:ccd1:8f8e:ffad:96b
WSDD:  + fd13:4514:2d3b::4cb
WSDD:  + 192.168.1.246
WSDD:  + fe80::d6f0:d5e:307e:3876%2
WSDD: 2a00:62c0:2be8:8400::4cb: started discovery, UDP port=35150
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: 2a00:62c0:2be8:8400:aaf8:2303:5cb0:8dc: started discovery, UDP port=55861
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: 2a00:62c0:2be8:8400:b809:d6e1:3fad:8460: started discovery, UDP port=50641
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: fd13:4514:2d3b:0:865b:3b22:f116:7e3c: started discovery, UDP port=58182
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: fd13:4514:2d3b:0:ccd1:8f8e:ffad:96b: started discovery, UDP port=40428
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: fd13:4514:2d3b::4cb: started discovery, UDP port=41166
WSDD: 192.168.1.246: started discovery, UDP port=57254
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: fe80::d6f0:d5e:307e:3876: started discovery, UDP port=52602
API: sane_get_devices(): called
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
zeroconf: found urn:uuid:72c7beb6-7042-4d96-bf8f-aeca8f9452f0
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 2 (enp34s0)
zeroconf:   name:      -
zeroconf:   model:     -
zeroconf:   addresses:
zeroconf:   protocol:  WSD
zeroconf:   endpoints:
zeroconf:   device:    0000 (created)
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: device_list wait: waiting for any proto for 'urn:uuid:72c7beb6-7042-4d96-bf8f-aeca8f9452f0' (0)
WSDD: 192.168.1.246: done discovery
WSDD: 2a00:62c0:2be8:8400::4cb: done discovery
WSDD: fd13:4514:2d3b:0:ccd1:8f8e:ffad:96b: done discovery
WSDD: fd13:4514:2d3b:0:865b:3b22:f116:7e3c: done discovery
WSDD: 2a00:62c0:2be8:8400:b809:d6e1:3fad:8460: done discovery
WSDD: fd13:4514:2d3b::4cb: done discovery
WSDD: fe80::d6f0:d5e:307e:3876: done discovery
WSDD: 2a00:62c0:2be8:8400:aaf8:2303:5cb0:8dc: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: device_list wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: urn:uuid:72c7beb6-7042-4d96-bf8f-aeca8f9452f0 (0): can: none, use: none
zeroconf: urn:uuid:72c7beb6-7042-4d96-bf8f-aeca8f9452f0 (0): skipping, none of supported protocols discovered
zeroconf: zeroconf_device_list_get: resulting list:
API: sane_get_devices(): done
device `lexmark_nscan:libnet/SPECIFY_DEVICE' is a Lexmark Network Scanner
device `lexmark_nscan:libnet/0021B7B1B79C' is a Lexmark Lexmark MX310dn Ethernet Scanner
API: sane_exit(): called
zeroconf: device gone urn:uuid:72c7beb6-7042-4d96-bf8f-aeca8f9452f0
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 2 (enp34s0)
API: sane_exit(): OK
```

aleksey1831981, 2023-11-02:
[Lexmark Ethernet simple-scan-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/13244667/Lexmark.Ethernet.simple-scan-zeroconf.log)
[Lexmark_Network simple-scan-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/13244669/Lexmark_Network.simple-scan-zeroconf.log)

aleksey1831981, 2023-11-02:
airscan-discover по прежнему ничего не видит.

```
$ airscan-discover
[devices]
```

alexpevzner, 2023-11-02:
Это устройство говорит в WS-Discovery, что оно принтер, но не сканер. Это может означать одно из двух:
* или устройство действительно не поддерживает WS-Scan и при этом поддерживает WS-Print (такое бывает)
* или WS-Scan выключен где-то в настройках

Я поискал в мануале ручку, выключающую WS-Scan, и не нашел. Но у меня нет самого устройства, есть только мануал. Попробуйте вы тоже поискать.

Это может называться не словом WSD, а "разрешить сканирование с компьютера по сети" или что-нибудь в этом роде.

aleksey1831981, 2023-11-02:
Я не нашел подобных опций в настройках.
Но я подключился к сканеру из винды на виртуальной машине. Устройство установилось сразу же. Сканирует по сети исправно. Никаких дополнительных настроек на устройстве я не совершал. В свойствах оборудования, указан драйвер WSDscan. Скрин во вложении. Более того на сайте Lexmark указано, что WSD поддерживается "Продукты серий MX, CX, XM, XC (поддержка сканирования и печати)".

https://infoserve.lexmark.com/ids/ifc/ids_topic.aspx?root=kb20211110015949938&topic=HO3780&productCode=Lexmark_MX310&loc=en_US

[image]

alexpevzner, 2023-11-02:
Да, вижу, что я не учел. Ну ОК, ждите обновления. Наверное, починю (сообщу, когда сделаю или потребуется еще какая-то помощь).

На всякий случай, вы сможете, если что, самостоятельно собрать sane-airscan с целью попробовать?

alexpevzner, 2023-11-02:
Не, я углядел, где он себя как сканер анонсирует. Просто он это делает не так, как все. Починю

aleksey1831981, 2023-11-03:
Хорошо. Буду ждать

aleksey1831981, 2023-11-06:
Александр, приветствую.
Зашел полюбопытствовать как дела с моим сканером? Не хочу показаться навязчивым, но на текущий момент для меня это важно.

alexpevzner, 2023-11-07:
Сможете сами собрать?

alexpevzner, 2023-11-07:
Я тут немного подкрутил. Попробуйте пересобрать.

Я обычно не делаю make install, а просто заменяю libsane-airscan.so.1 из установленного пакета ссылкой на libsane-airscan.so.1 из моей директории - этого достаточно для корректной работы, и при внесении изменений позволяет видеть результат сразу после пересборки (нет нужды каждый раз переставлять).

aleksey1831981, 2023-11-07:
Да, конечно. Откуда нужно скачать libsane-airscan.so.1 ?

aleksey1831981, 2023-11-07:
https://software.opensuse.org//download.html?project=home%3Apzz&package=sane-airscan

aleksey1831981, 2023-11-07:
Александр, моё почтение. Сканер работает. Автоподатчик тоже работает. Ошибок нет. 
Магия

aleksey1831981, 2023-11-07:
Только вот `airscan-discover` почему то ничего не показывает

alexpevzner, 2023-11-07:
> Только вот airscan-discover почему то ничего не показывает

Потому, что это тот, который у вас в пути, старый, пришедший с пакетом. `./airscan-discover` в сборочной директории должен все запоказывать.

aleksey1831981, 2023-11-08:
Сработало.
Спасибо еще раз за работающий сканер в отдельно взятой семье в частности, и за вклад в развитие свободного общества в целом

aleksey1831981, 2023-11-08:
Александр, пришел с обновленной информацией.
Оказывается этот сканер умеет еSCL.
Почему то в веб настройках я пропустил нужный пункт
Параметры -> Сеть/порты -> AirPrint
Там нужно поставить галочку (чекбокс) "Вкл". Примечательно, что после того, как поставить эту галку в настройках и перезагрузки устройства, данный чекбокс пропадает. Т.е. выключить уже нельзя.

Теперь `airscan-discover`, старый, пришедший с пакетом по умолчанию, видит устройство eSCL сразу.
```
:~$ airscan-discover
[devices]
  MX310dn = http://192.168.1.243:80/eSCL/, eSCL
``` 

Тот, который вы пофиксили, скачанный отсюда, видит и WSD IPv4, WSD IPv6, и eSCL.
```
:~/sane-airscan$ ./airscan-discover
[devices]
  MX310dn = http://192.168.1.243:80/eSCL/, eSCL
  MX310dn = http://[2a00:62c0:2b5a:5300::243]:65004/Scanner1/WebServices, WSD
  MX310dn = http://192.168.1.243:65004/Scanner1/WebServices, WSD
``` 

Функционал проверю завтра. Сегодня уже поздно. Если все работает корректно в eSCL - отпишусь.

aleksey1831981, 2023-11-09:
Докладываю.
eSCL тоже работает исправно. 
С планшета сканирует, из АПД сканирует, пустой лоток видит, из спящего режима выходит.