alexpevzner/sane-airscan #305: Compile fail with git commit 57b2bef3d5f4f06

gene-git, 2023-12-02:
You presumably already aware, but our daily test builds started failing - this is as of commit 57b2bef3d5f4f06

gene

```
gcc -c -o objs/airscan-xml.o airscan-xml.c  -march=x86-64-v3 -mtune=generic -O2 -pipe -fno-plt -fexceptions        -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security        -fstack-clash-protection -fcf-protection -std=gnu18 -O2 -g -W -Wall -Werror -pthread  -D CONFIG_SANE_CONFIG_DIR=\"/etc/sane.d\"  -D_REENTRANT  -I/usr/include/libxml2  -I/usr/include/p11-kit-1    -I/usr/include/libpng16   -fPIC
airscan-xml.c: In function ‘xml_rd_parse’:
airscan-xml.c:126:23: error: assignment to ‘xmlStructuredErrorFunc’ {aka ‘void (*)(void *, const struct _xmlError *)’} from incompatible pointer type ‘void (*)(void *, xmlError *)’ {aka ‘void (*)(void *, struct _xmlError *)’} [-Werror=incompatible-pointer-types]
  126 |     ctxt->sax->serror = xml_rd_error_callback;
      |                       ^
airscan-xml.c: In function ‘xml_rd_node_value_uint’:
airscan-xml.c:406:9: error: implicit declaration of function ‘strtoul’; did you mean ‘strtok’? [-Werror=implicit-function-declaration]
  406 |     v = strtoul(s, &end, 10);
      |         ^~~~~~~
      |         strtok
cc1: all warnings being treated as errors
make: *** [Makefile:100: objs/airscan-xml.o] Error 1
==> ERROR: A failure occurred in build().
    Aborting...
```

alexpevzner, 2023-12-02:
Hi @gene-git,

strange if this problem caused by 57b2bef3d5f4f06, because that commit doesn't touch anything related to XML. It looks more similar to libxml header mismatch between your distro and mine.

The second error (implicit declaration of the `strtoul`) is easy to fix. It is caused by missed `#include <stdlib.h>`, which is implicitly included from some library headers that I use on most distros, but not on yours.

What system do you build for?

gene-git, 2023-12-02:
Yeh sorry - didn't mean to imply that commit caused anything - just the build pulled git HEAD and that was the commit - builds stopped working I think last night so probably any commit from last day or 2.

This could be triggered by the gcc build flags specially perhaps  -Werror (turning all warnings into errors).

```
   cc ... -march=x86-64-v3 -mtune=generic -O2 -pipe -fno-plt -fexceptions        -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security        -fstack-clash-protection -fcf-protection -std=gnu18 -O2 -g -W -Wall -Werror -pthread  ...
```

This is Arch linux using gcc version :  13.2.1

alexpevzner, 2023-12-02:
In my `libxml2/libxml/xmlerror.h` (libxml2-2.10.4-1.fc38.x86_64):

```
typedef void (XMLCALL *xmlStructuredErrorFunc) (void *userData, xmlErrorPtr error);
```

In the current libxml2 git:

```
typedef void (*xmlStructuredErrorFunc) (void *userData, const xmlError *error);
```

The difference in the word **const** which was added to the second parameter in the recent versions of libxml2.

Could you please try to add an explicit cast to the `sane-airscan` sources? `airscan-xml.c, line 127` should look like:

```
    ctxt->sax->serror = (xmlStructuredErrorFunc) xml_rd_error_callback;
```

I want you to try before adding it to the repo, because not 100% sure that your gcc will eat it (even the same versions behave sometimes differently between distros)

gene-git, 2023-12-02:
Okay will do. will report back.

gene-git, 2023-12-02:
I confirm - the cast does the trick - build completes fine with it.

alexpevzner, 2023-12-02:
OK. If it helps. we can fix this problem very quickly. Otherwise, oh... (it will be different between distros)

alexpevzner, 2023-12-02:
What is the Arch version (for change comment)?

gene-git, 2023-12-02:
I've added a rebuild trigger on any change to libxml2 - so if something else changes that trips something, will get notified sooner next time.

Thanks kindly for figuring it out and resolving so quickly

alexpevzner, 2023-12-02:
Commited.

Adding #ifdef has one difficulty: it is hard to guess the exact condition to put under that #ifdef.

commit 144e7d41 (2023-12-02): Fixed build with recent versions of libxml2 (fixes #305)

libxml2 recently changed a prototype of the xmlStructuredErrorFunc callback,
adding const modifier to the second parameter.

Adding an explicit cast seems to fix the problem (but I will not be surprised, if
this fix will not work equally well on all distros).

Found for a first time and fixed on Arch linux on about 2 Dec 2023 (Arch is rolling
release, so there is no explicit release number)