alexpevzner/sane-airscan #307: Regression for HP OfficeJet 6950 and HP OfficeJet 6960

essembeh, 2023-12-08:
Hi

It 's been month since my scanner (Wireless HP OfficeJet 6950) does not work anymore (I was able to reproduce the same behavior with a HP OfficeJet 6960).

I use NixOS and `simple-scan` was working fine in NixOS 22.11. Since 23.05 (and for 23.11), I get an error "Document Feeder empty" whereas  I selected scan from flatbed.

After some investigations, I was able to use it by downgrading the flatpak version of "Simple Scan".

Here is the result of `simple-scan --debug`
```
[+0.00s] DEBUG: simple-scan.vala:2015: Starting simple-scan 44.0, PID=25091
[+0.00s] DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
[+0.14s] DEBUG: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
[+0.17s] DEBUG: app-window.vala:2002: Loading state from /home/seb/.config/simple-scan/state
[+0.17s] WARNING: scanner.vala:1739: Unknown ScanType: . Please report this error.
[+0.17s] DEBUG: app-window.vala:1981: Restoring window to 600x400 pixels
[+0.19s] DEBUG: scanner.vala:1619: sane_init () -> SANE_STATUS_GOOD
[+0.19s] DEBUG: scanner.vala:1625: SANE version 1.2.1
[+0.19s] DEBUG: scanner.vala:1686: Requesting redetection of scan devices
[+0.19s] DEBUG: scanner.vala:863: Processing request
[+0.45s] DEBUG: app-window.vala:2078: Saving state to /home/seb/.config/simple-scan/state
[+8.74s] DEBUG: scanner.vala:348: sane_get_devices () -> SANE_STATUS_GOOD
[+8.74s] DEBUG: scanner.vala:374: Device: name="escl:https://10.0.2.2:443" vendor="HP" model="OfficeJet 6950 [CFC9A1]" type="platen,adf scanner"
[+9.05s] DEBUG: app-window.vala:2078: Saving state to /home/seb/.config/simple-scan/state
[+9.73s] DEBUG: app-window.vala:2078: Saving state to /home/seb/.config/simple-scan/state
[+10.15s] DEBUG: app-window.vala:2078: Saving state to /home/seb/.config/simple-scan/state
[+10.27s] DEBUG: simple-scan.vala:1820: Requesting scan at 300 dpi from device 'escl:https://10.0.2.2:443'
[+10.27s] DEBUG: scanner.vala:1777: Scanner.scan ("escl:https://10.0.2.2:443", dpi=300, scan_mode=ScanMode.COLOR, depth=8, type=single, side=front, paper_width=0, paper_height=0, brightness=0, contrast=0, delay=0ms)
[+10.27s] DEBUG: scanner.vala:863: Processing request
[+10.59s] DEBUG: app-window.vala:2078: Saving state to /home/seb/.config/simple-scan/state
[+11.13s] DEBUG: scanner.vala:924: sane_open ("escl:https://10.0.2.2:443") -> SANE_STATUS_GOOD
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (0)
[+11.13s] DEBUG: scanner.vala:795: Option 0: name='(null)' title='Number of options' type=int size=4 cap=soft-detect
[+11.13s] DEBUG: scanner.vala:798:   Description: Read-only option that specifies how many options a specific device supports.
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (1)
[+11.13s] DEBUG: scanner.vala:795: Option 1: name='(null)' title='Scan mode' type=group size=0
[+11.13s] DEBUG: scanner.vala:798:   Description: 
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (2)
[+11.13s] DEBUG: scanner.vala:795: Option 2: name='mode' title='Scan mode' type=string size=6 values=["Gray", "Color"] cap=soft-select,soft-detect
[+11.13s] DEBUG: scanner.vala:798:   Description: Selects the scan mode (e.g., lineart, monochrome, or color).
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (3)
[+11.13s] DEBUG: scanner.vala:795: Option 3: name='resolution' title='Scan resolution' type=int size=4 unit=dpi values=[75, 100, 200, 300, 600, 1200] cap=soft-select,soft-detect
[+11.13s] DEBUG: scanner.vala:798:   Description: Sets the resolution of the scanned image.
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (4)
[+11.13s] DEBUG: scanner.vala:795: Option 4: name='source' title='Scan source' type=string size=8 values=["Flatbed", "ADF"] cap=soft-select,soft-detect
[+11.13s] DEBUG: scanner.vala:798:   Description: Selects the scan source (such as a document-feeder).
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (5)
[+11.13s] DEBUG: scanner.vala:795: Option 5: name='(null)' title='Geometry' type=group size=0 cap=advanced
[+11.13s] DEBUG: scanner.vala:798:   Description: Scan area and media size options
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (6)
[+11.13s] DEBUG: scanner.vala:795: Option 6: name='tl-x' title='Top-left x' type=fixed size=4 unit=mm min=0.000000, max=215.222656, quant=0 cap=soft-select,soft-detect
[+11.13s] DEBUG: scanner.vala:798:   Description: Top-left x position of scan area.
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (7)
[+11.13s] DEBUG: scanner.vala:795: Option 7: name='tl-y' title='Top-left y' type=fixed size=4 unit=mm min=0.000000, max=299.889328, quant=0 cap=soft-select,soft-detect
[+11.13s] DEBUG: scanner.vala:798:   Description: Top-left y position of scan area.
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (8)
[+11.13s] DEBUG: scanner.vala:795: Option 8: name='br-x' title='Bottom-right x' type=fixed size=4 unit=mm min=0.677322, max=215.899994, quant=0 cap=soft-select,soft-detect
[+11.13s] DEBUG: scanner.vala:798:   Description: Bottom-right x position of scan area.
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (9)
[+11.13s] DEBUG: scanner.vala:795: Option 9: name='br-y' title='Bottom-right y' type=fixed size=4 unit=mm min=0.677322, max=300.566666, quant=0 cap=soft-select,soft-detect
[+11.13s] DEBUG: scanner.vala:798:   Description: Bottom-right y position of scan area.
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (10)
[+11.13s] DEBUG: scanner.vala:795: Option 10: name='(null)' title='Enhancement' type=group size=0 cap=advanced
[+11.13s] DEBUG: scanner.vala:798:   Description: 
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (11)
[+11.13s] DEBUG: scanner.vala:795: Option 11: name='preview' title='Preview' type=bool size=4 cap=soft-select,soft-detect
[+11.13s] DEBUG: scanner.vala:798:   Description: Request a preview-quality scan.
[+11.13s] DEBUG: scanner.vala:1002: sane_get_option_descriptor (12)
[+11.13s] DEBUG: scanner.vala:795: Option 12: name='preview-in-gray' title='Force monochrome preview' type=bool size=4 cap=soft-select,soft-detect
[+11.13s] DEBUG: scanner.vala:798:   Description: Request that all previews are done in monochrome mode.  On a three-pass scanner this cuts down the number of passes to one and on a one-pass scanner, it reduces the memory requirements and scan-time of the preview.
[... 17 lines cut ...]
[+11.13s] DEBUG: scanner.vala:584: sane_control_option (9, SANE_ACTION_SET_VALUE, br-y=option.range.max=300.566666) -> (SANE_STATUS_GOOD)
[+11.13s] DEBUG: scanner.vala:449: scale_int: scaling 0 [min: -100, max: 100] to 1000 [min: 0, max: 2000]
[+11.13s] DEBUG: scanner.vala:514: sane_control_option (13, SANE_ACTION_SET_VALUE, brightness=1000) -> (SANE_STATUS_GOOD, 1000)
[+11.13s] DEBUG: scanner.vala:449: scale_int: scaling 0 [min: -100, max: 100] to 1000 [min: 0, max: 2000]
[+11.13s] DEBUG: scanner.vala:514: sane_control_option (14, SANE_ACTION_SET_VALUE, contrast=1000) -> (SANE_STATUS_GOOD, 1000)
[+12.25s] DEBUG: scanner.vala:1352: sane_start (page=0, pass=0) -> SANE_STATUS_NO_DOCS
[+12.25s] DEBUG: scanner.vala:1325: sane_cancel ()
[+12.25s] DEBUG: scanner.vala:825: sane_cancel ()
[+12.25s] DEBUG: scanner.vala:828: sane_close ()
[+12.25s] CRITICAL: page_get_has_data: assertion 'self != NULL' failed
[+12.25s] CRITICAL: book_delete_page: assertion 'page != NULL' failed
[+12.61s] DEBUG: app-window.vala:2078: Saving state to /home/seb/.config/simple-scan/state
[+13.80s] DEBUG: app-window.vala:2078: Saving state to /home/seb/.config/simple-scan/state
[+14.43s] DEBUG: app-window.vala:2078: Saving state to /home/seb/.config/simple-scan/state
[+15.04s] DEBUG: autosave-manager.vala:201: Deleting autosave records
[+15.04s] WARNING: autosave-manager.vala:214: Failed to delete autosaves: Error opening directory “/home/seb/.cache/simple-scan/autosaves”: No such file or directory
[+15.04s] DEBUG: scanner.vala:1807: Stopping scan thread
[+15.04s] DEBUG: scanner.vala:863: Processing request
[+15.05s] DEBUG: scanner.vala:1818: sane_exit ()
```

And here is a capture of what happens:
[image]

And here is the post I initially opened on [NixOS forum](https://discourse.nixos.org/t/help-with-a-wireless-scanner/36439)

alexpevzner, 2023-12-09:
Hi @essembeh,

> DEBUG: scanner.vala:924: sane_open ("escl:https://10.0.2.2:443") -> SANE_STATUS_GOOD

We actually have two independent eSCL drivers for SANE: `sane-escl` and `sane-airscan`. In your case, `simple-scan` uses `sane-escl` (sane_open ("_**escl**_:https://10.0.2.2:443")). You are writing issue against `sane-airscan`.

So my suggestion would be to try to switch to `sane-airscan`. If it wouldn't help, you are welcome to file a separate issue.

closed 2023-12-09 by alexpevzner

essembeh, 2023-12-10:
Hi @alexpevzner 

Thanks for the details, I'll open a ticket on sane-escl project.
May I ask some more questions, as this is rather obscure to me.

How can I switch to sane-airscan instead of sane-escl?
Is that possible with simple-scan to force using another driver?
There is only one choice in simple-scan selection list...

Thanls

essembeh, 2024-01-08:
Just for information, 
I was finally able to use airscan driver instead of the escl driver for my HP 6950.

I had to explicitly disable `escl` so that sane uses `airscan` to scan ... and it works.

There is still an open ticket on `sane-backends` project because the `escl` driver does not work. But thanks `airscan`, I can scan :) 

Thanks