alexpevzner/sane-airscan #313: brother MFC-J485DW please help with debugging:

panorain, 2024-01-06:
I am sorry if this is not the correct place to post the following message. I have been trying to send email to pzz@apevzner.com and it fails. Can you please tell me how to contact with email so I can attach the .tar file?

Hi, operating machine under openSUSE Tumbleweed fully updated, sane-airscan. Device is model brother MFC-J485DW . brother Driver Install Tool installed with wireless i.p. address 192.168.1.20 xsane works, printing works. Please help with debugging of sane-airscan.  Using starter (Alt+f2), open xsane then selecting eSCL fails (2 screenshots attached to email also).  I have attached a screenshot to show sane-airscan repository is enabled for reference if needed. Also have attached 1 of the 2 requested files for debugging help. I have verified before that the brother MFC-J485DW sane-airscan eSCL mode works on LinuxMint so hopefully I can get assistance to verify again eSCL and even WSD on brother MFC-J485DW on openSUSE Tumbleweed.

Attached is 1 of the 2 files required for help with debugging. Obviously I am doing something wrong with sane-airscan email address. 
 
**Note** I have recently verified that brother **MFC-J475DW** does work with WSD (Linux Miint 20.2). Did not debug that one yet (different location) so not certain if eSCL is supported on it.
 
Thank you for sane-airscan

[image]
Since now I have disabled autorefresh on the sane-airscan repository.
[image]
[image]
[xsane-brother-MFC-J485DW.log](https://github.com/alexpevzner/sane-airscan/files/13849851/xsane-brother-MFC-J485DW.log)

alexpevzner, 2024-01-07:
Hi @panorain,

pzz@apevzner.com is the valid e-mail address. Probably, it was some temporary delivery error, or e-mail message was too large. You can also try my alternative e-mail, pzz@pzz.msk.ru

Looking to the log, I see absolutely no errors there. Have you tried any other SANE client (`simple-scan`, for example)?

panorain, 2024-01-08:
Thank you for your response. What I did was change /etc/sane.d/airscan.conf as follows on the 2 machines here, to use port 80 and not port 9050. I do hope that pasting the current contents of /etc/sane.d/airscan.conf is ok with you to review. If it is ok with you to review the file please do and any thoughts on ports are most welcome also. Thanks again for the prompt response and wonderful program.

 

Thinkcentre-M57p:/etc/sane.d> cat airscan.conf 
# sane-airscan example configuration file

# You can add scanners manually, using the following syntax:
# [devices]
#   name1 = url1           ; add eSCL device
#   name2 = url2, protocol ; protocol can be escl or wsd
#   name2 = disable        ; disable the device
#
# The following utility helps to discover scanners for manual
# addition:
#
#   https://github.com/alexpevzner/airscan-discover

[devices]
#"brother-MFC-J485DW" = http://192.168.1.20:8080/eSCL
"brother-MFC-J485DW" = http://192.168.1.20:80/eSCL
#"Kyocera MFP Scanner" = http://192.168.1.102:9095/eSCL
#"Some Unwanted Scanner" = disable

# Various options
#
# Automatic discovery may be enabled (the default) or disabled:
#   discovery = enable  ; Enable automatic device discovery
#   discovery = disable ; Disable both DNS-SD and WS-Discovery
#
# Protocol choice (eSCL vs WSD if both are available)
#   protocol = auto     ; Best protocol is chosen automatically, the default
#   protocol = manual   ; Manual choice is offered
#
# WS-Discovery mode
#   ws-discovery = fast ; Fast discovery, the default
   ws-discovery = full ; Full discovery, slow and accurate
#   ws-discovery = off  ; Disable WS-Discovery
#
# Scanner "model" is a string that most of SANE apps display in a list
# of devices. It may be more convenient to use scanner network name
# for this purpose:
#   model = network     ; use network device name (default)
#   model = hardware    ; use hardware model name
#
# socket_dir gives an optional path to a directory where local (UNIX) sockets
# can be found.  If an eSCL device's URL is in the form unix://socket/eSCL/,
# traffic will be sent through socket_dir/socket instead of TCP.  If not
# specified, sockets will be searched for in /var/run.

[options]
#discovery = enable
#model = network
#protocol = auto
#ws-discovery = fast
#socket_dir = /var/run

# Configuration of debug facilities
#   trace = path         ; enables protocol trace and configures output
#                        ; directory. The directory will be created
#                        ; automatically. Path may start with tilde (~)
#                        ; character, which means user home directory
#
#   enable = true|false  ; enable or disable console logging
#   hexdump = true|false ; hex dump all traffic (very verbose!)
[debug]
#trace   = ~/airscan/trace
#enable  = true
#hexdump = false

# Blacklisting devices
#   model = pattern     ; Blacklist devices by model name
#   name  = pattern     ; Blacklist devices by network name
#   ip    = addr[/mask] ; Blacklist devices by IP addresses
#
# Notes
#   In model and network names glob-style wildcards can be used
#   (i.e., model = "Xerox*")
#
#   Network names come from DNS-SD, WS-Discovery doesn't provide this
#   information. For filtering by network name to work, Avahi must be
#   enabled and device must be discoverable via DNS-SD (not necessarily
#   as a scanner, it's enough if WSD scanner is discoverable as a printer
#   via DNS-SD).
#
#   If netmask is not set, address assumed to be device address,
#   not address of the entire subnet
#
#   Blacklisting only affects automatic discovery, and doesn't
#   affect manually configured devices
[blacklist]
#model = "Xerox*"       ; blacklist by model name
#name  = "HP*"          ; blacklist by network name
#ip    = 192.168.0.1    ; blacklist by address
#ip    = 192.168.0.0/24 ; blacklist the whole subnet

[debug]
trace = ~/airscan/trace ; Path to directory where trace files will be saved
Thinkcentre-M57p:/etc/sane.d> 

 

 
 

Sent: Sunday, January 07, 2024 at 3:11 PM

 

Hi @panorain,

panorain, 2024-01-08:
> Hi @panorain,
>
> different devices use different TCP ports for eSCL. Kyocera included only as example, and though Kyocera actually uses 9095 port, most of devices I've ever seen ise 80 or 8080. This stuff is designed with automatic discovery in mind, so typically user should not bother about port number and similar things.
>
> Why do you need to manually specify the device? Automatic discovery doesn't work for you, or you don't like Avahi?

Hi, Alexander

 

I see now more about the port numbers. I can say with the MFC-J485DW the following: After downloading and then executing the 'Driver Install Tool' from the brother website. Opening XSANE automatically detects the devices scanner as 'net;dev0'. The initial reason of the use of sane-airscan was for me dealing with the MFC-J485DW in Linux-Mint 19.3 XSANE not being able to detect the device (after upgrade from 19.1 to 19.3 there was some sort of regression I believe. I then reached out on the sane-devel-list and was brought to the attention of sane-airscan then after installing the package I was able to use the devices scanner after minor modification to the 'airscan.conf' file. It was a while ago and very details are not extreme clear right now (I can look up some things if needed at all). The reason for my attempts with sane-airscan on the 3 systems currently here are for more options of scanning really I think.

 

1. Are you meaning by "Automatic discovery" brother-MFC-J485DW is detected after installation of the brother driver package for the device?

2. I do not myself know about Avahi and how it is a part with the scanning.

3. Each time in order to use sane-airscan after installation I add the following entry line to the airscan.conf file under the [devices] category as follows for this device:  "brother-MFC-J485DW" = http://192.168.1.20:80/eSCL then XSANE offers the eSCL sane-airscan option as well as default brother scanning option. I have attached a screenshot to this email of the currently offered options proveded by XSANE now which is the same on each of the 3 machines here now. 1 machine Linux Mint 21.2, two machines openSUSE Tumbleweed. 

4. Is it safe to say that the brother-MFC-485DW does not support WSD? Or is an adjustment or more testing involved to say this?

5. Are you aware of a way to work with sane-airscan for Amazon Kindle or Amazon Fire at all?

 

Thanks for helping make this more clear to me.

Sent: Monday, January 08, 2024 at 4:03 AM