alexpevzner/sane-airscan #314: ADF source fails with HP ScanJet Pro 2000 s2

gabrieleturchi, 2024-01-13:
When scanning with scanimage --source "ADF Duplex" everything works fine, but without --source options, or with plain ADF argument (the default), the image appear to be a narrow and very long grey bar.

alexpevzner, 2024-01-16:
Hi @gabrieleturchi,

can I see protocol trace?

In the `/etc/sane.d/airscan.conf` uncomment the following lines:

```
[debug]
trace = ~/airscan/trace
```

Trace files will be in the `~/airscan/trace ` directory which will be created automatically. `.tar` files are not needed for now, `.log` files would be enough

gabrieleturchi, 2024-01-22:
Here the traces.

[scanimage-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/14009777/scanimage-zeroconf.log)
[scanimage-HP-ScanJet-Pro-2000-s2-(USB).log](https://github.com/alexpevzner/sane-airscan/files/14009778/scanimage-HP-ScanJet-Pro-2000-s2-.USB.log)

alexpevzner, 2024-01-23:
Could you please test with the following options: `scanimage -y 355.6mm`

gabrieleturchi, 2024-01-23:
It works. I suppose that this driver/scanner doesn't support the auto detection of the end of the page...

alexpevzner, 2024-01-23:
What if you insert A4 page and request scan with `scanimage -y 700mm`?

Or, better, could you play a little bit around the `-y` option trying to find situation where the scanning breaks?

alexpevzner, 2024-01-28:
Ping

alexpevzner, 2024-01-28:
Hi @gabrieleturchi,

I'm about to release next version of `sane-airscan` within few days.

Your device reports max image size 215.90x**3098.80** mm when used in the non-duplex ADF mode (please notice the height). Probably it means it can scan not only pages, but continuous documents as well.

But image of that size is actually requested for scanning (what scanimage does, if size is not explicitly specified) it returns whatever you see.

Probably if I artificially limit image height for your device in the driver, everything will work as expected.

However I need your help here to figure out the reasonable heigh limit. I don't have my own access to such a device.

Thanks in advance!

gabrieleturchi, 2024-02-02:
Hi alex. Sorry, my first report was not precise: in fact, the narrow grey bar had on the top the image of the page... (shame on eog who showed only the grey background...). It is the scan of a 3 meters long page...  Apparently, in non-duplex mode the scanner read for the full height, when in duplex mode it detects the end of the paper (not very precise, I can see at least 10% extra height in grey, but a different grey of the background).

alexpevzner, 2024-02-06:
Looks that the problem actually doesn't exist. I've also asked a colleague who has a similar device to retest just in case, and colleague claims that everything is OK for him.

So closing the issue...

closed 2024-02-06 by alexpevzner

gabrieleturchi, 2024-02-06:
So, you confirm that there is no way to detect the end of the page in single side mode? Not very handy... I suggest to implement a default y size (if not specified by the user) around 33 cm. In any case, a notice in the documentation could be useful, I think.

Thank you.