alexpevzner/sane-airscan #316: saned is not show Airscan scanners

zekzit, 2024-01-26:
I have use `scanimage -L` it show devices that prefix with `airscan:<scanner>`. But when I enable net backend, it show only device that prefix with `net:localhost:<scanner>` but not `net:localhost:airscan:<scanner>`. So, how can I share this airscan scanner through saned? Or it cannot be shared?

alexpevzner, 2024-01-27:
Hi @zekzit,

this is by design of `saned`. When SANE "frontend" asks its "backends" for a list of connected devices, using the `sane_get_devices` function, it can specify via its second parameter, if it is interested in all devices or locally connected only ones.

As eSCL/WSD devices are all network devices, i.e. not locally-connected, `sane-airscan` returns empty list when only local devices are requested.

I've briefly reviewed another backends, they behave on a similar manner.

I wonder why somebody may want to re-export network device via `saned` instead of connecting to it directly.

But anyway, if you feel that `saned` behaviour should be somehow changed, it is better to file issue into the saned repository at GitLab: https://gitlab.com/sane-project/backends

alexpevzner, 2024-01-27:
I will convert this conversation to discussion, as it is not sane-airscan issue actually...