alexpevzner/sane-airscan #321: Please add HP Color LaserJet Enterprise Flow MFP M880z & M880z+

guiodic, 2024-02-01:
According https://support.apple.com/en-us/HT201311 it is airscan compatible.

Thank you.

guiodic, 2024-02-01:
airscan-discover -d gives this output but airscan-discover without -d seems find nothing and sane apps doesnt' find the scanner

MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 192.168.1.117@enp0s20f0u2u4, ifindex=7
WSDD:  + fe80::ce1b:46ad:d00d:7fe1@enp0s20f0u2u4, ifindex=7
WSDD: 192.168.1.117: started discovery, UDP port=36449
WSDD: fe80::ce1b:46ad:d00d:7fe1: started discovery, UDP port=50469
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv4@enp0s20f0u2u4("_ipp._tcp"): AVAHI_BROWSER_NEW (CACHED MCAST)
MDNS: browse-ipv4@enp0s20f0u2u4("_ipp._tcp"): "HP Color LaserJet flow MFP M880 [177BC2]" (CACHED MCAST)
MDNS: browse-ipv6@*("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv4@enp0s20f0u2u4("_ipps._tcp"): AVAHI_BROWSER_NEW (CACHED MCAST)
MDNS: browse-ipv4@enp0s20f0u2u4("_ipps._tcp"): "HP Color LaserJet flow MFP M880 [177BC2]" (CACHED MCAST)
MDNS: browse-ipv6@*("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6@*("_uscan._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6@*("_uscans._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6@*("_scanner._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: resolve-ipv4@enp0s20f0u2u4("_ipp._tcp", "HP Color LaserJet flow MFP M880 [177BC2]"): 192.168.1.200:631 (CACHED MCAST)
WSDD: directed probe: trying if=enp0s20f0u2u4, addr=192.168.1.200
WSDD: HTTP POST http://192.168.1.200/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
WSDD: HTTP resolving 192.168.1.200 80
WSDD: HTTP trying 192.168.1.200:80
WSDD: HTTP fdpoll: {--} -> {-w}
WSDD: HTTP 804 bytes sent
WSDD: HTTP done request sending
WSDD: HTTP fdpoll: {-w} -> {rw}
WSDD: HTTP fdpoll: {rw} -> {r-}
MDNS: resolve-ipv4@enp0s20f0u2u4("_ipps._tcp", "HP Color LaserJet flow MFP M880 [177BC2]"): 192.168.1.200:631 (CACHED MCAST)
WSDD: directed probe: trying if=enp0s20f0u2u4, addr=192.168.1.200
WSDD: directed probe: already in progress, skipping
zeroconf: found urn:uuid:f6060e84-ff8d-d4a4-64ae-9cb654177bc2
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 7 (enp0s20f0u2u4)
zeroconf:   name:      -
zeroconf:   model:     -
zeroconf:   addresses:
zeroconf:   protocol:  WSD
zeroconf:   endpoints:
zeroconf:   device:    0000 (created)
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6@*("_ipp._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6@*("_ipps._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6@*("_uscan._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
MDNS: browse-ipv6@*("_uscans._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
MDNS: browse-ipv6@*("_scanner._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: device_list wait: DNS-SD not finished...
WSDD: HTTP 477 bytes received
WSDD: HTTP POST http://192.168.1.200/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: got response headers (200)
WSDD: HTTP 2613 bytes received
WSDD: HTTP 5 bytes received
WSDD: HTTP done response reception
WSDD: HTTP POST http://192.168.1.200/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: 200 OK
WSDD: HTTP POST http://192.168.1.200/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: OK
WSDD: 192.168.1.117: done discovery
WSDD: fe80::ce1b:46ad:d00d:7fe1: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
zeroconf: initial scan timer expired
zeroconf: device_list wait: DNS-SD not finished...
zeroconf: device_list wait: timeout
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: urn:uuid:f6060e84-ff8d-d4a4-64ae-9cb654177bc2 (0): can: none, use: none
zeroconf: urn:uuid:f6060e84-ff8d-d4a4-64ae-9cb654177bc2 (0): skipping, none of supported protocols discovered
zeroconf: zeroconf_device_list_get: resulting list:
[devices]
zeroconf: device gone urn:uuid:f6060e84-ff8d-d4a4-64ae-9cb654177bc2
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 7 (enp0s20f0u2u4)
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished

alexpevzner, 2024-02-01:
Hi @guiodic,

can I see `airscan-discover -t` output (`airscan-discover-zeroconf.log`)?

guiodic, 2024-02-01:
[airscan-discover-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/14129054/airscan-discover-zeroconf.log)

alexpevzner, 2024-02-01:
Hi @guiodic,

looking to the log, I see no signs that this device supports any kind of wireless scanning, either eSCL or WSD. It announces IPP printing (the printing part of AirPrint) and WSD printing but doesn't announce scanning.

Experience shows that inclusion device into the Apple's list of AirPrint-compatible printers doesn't guarantee that it scans as well.

However some devices may have this support disabled by default but available via configuration. I've downloaded your device's manual but was not able to find there any recommendations how to enable these protocols. Honestly, scanner configuration is almost undocumented there.

So I can recommend you to browse device's configuration (web) console trying to find these controls by yourself. The keywords to look for may be "WSD", "eSCL", "AirPrint scanning", "Bonjour", "DNS-SD", "Enable scan from computer", "Scan from Mobile device", "Mopria scan", "AirScan".

Please let me know if you'll succeed, I'll update the README

closed 2024-02-01 by alexpevzner