alexpevzner/sane-airscan #326: Unable to connect to Brother ADS-2700W

pschichtel, 2024-02-21:
It worked with the previous version 0.99.27 and stopped working with 0.99.28. I've bisected the issue to cb8a28e601551d94c30a23e07d8277b0786f7fda.

airscan-discover output with the last commit that works:

```
[devices]
  Brother ADS-2700W [000ec6efc1f6] = http://192.168.1.154:53048, WSD
```

airscan-discover output with cb8a28e601551d94c30a23e07d8277b0786f7fda:

```
[devices]
  Brother ADS-2700W = http://192.168.1.154:53048, WSD

pschichtel, 2024-02-21:
My system is dual stack, but the scanner is connected only via IPv4.

pschichtel, 2024-02-21:
Just reverting that single commit fixes it.

alexpevzner, 2024-02-21:
Hi @pschichtel,

could you please run `airscan-discover -t` with and without that commit? It will create the `airscan-discover-zeroconf.log` file, I would ask you to publish both.

pschichtel, 2024-02-21:
@alexpevzner https://gist.github.com/pschichtel/a509f9bdafc765441816b77479124d96

alexpevzner, 2024-02-22:
Hi @pschichtel,

it's a quite clear what is broken in discovery and I will fix it soon.

What is not clear, is why device cannot be connected. Even with this problem discovery ends with the valid endpoint URL, so scanner must work.

Can I see a protocol trace?

pschichtel, 2024-02-22:
By protocol trace, do you mean something like tcpdump?

alexpevzner, 2024-02-22:
No. In the `/etc/sane.d/airscan.conf` uncomment the following lines:

```
[debug]
trace = ~/airscan/trace
```

Then try to scan. The `~/airscan/trace` directory will be created and the very long log files ("protocol traces") will be written there. I want to see such a file.

Only *.log files are interesting for now, *.tar files not required

pschichtel, 2024-02-22:
Sure, I can do that later today.

alexpevzner, 2024-02-22:
The problem in discovery was fixed - waiting for your confirmation that the fix works for you.

And I still want protocol trace to look to that part of the problem that I cannot understand/reproduce.

pschichtel, 2024-02-22:
yes, latest master works for me.

here is the trace logs: https://gist.github.com/pschichtel/7c892ab88ca43e7d9d509986001f7a3f

closed 2024-02-22 by pschichtel