alexpevzner/sane-airscan #327: Convenience vs. Speed: please!

yuv, 2024-02-26:
TLDR: don't force my system to waste my time on discovery every time.

Thank you for this great idea and for generally contributing to an easier configuration of scanning, and apology if the technical details in this report are a bit sparse.  This is first and foremost about user experience, and the principle:

1. Devices/network/configuration changes have a frequency of years: they happen rarely.
2. Scanning operations have a frequency of hours, sometimes even minutes: they happen multiple time each days. they happen more frequently than configuration changes.
3. Slowing the scanning operation by `scanimage` to do "automatic and manual device discovery and configuration" every time the user triggers a scan is a wasteful and bad default.

I don't know when the Ubuntu 22.04LTS system I am using as my daily driver upgraded, but at some point last week scanimage became painfully slow to start.  Running it with `SANE_DEBUG_DLL=5 scanimage -L` revealed the culprit.

`sudo apt purge sane-airscan` fixed it, and restored my regular productive speed.

Can you please add a pairing mode to the "automatic and manual device discovery and configuration" feature?

default in `/etc/sane.d/airscan.conf` would be
```
[options]
discovery = pair
```
when discovery is set to pair, do the automatic and manual device discovery and configuration and

1. automatically add the newly discovered devices to `[devices]`
2. for devices already configured that are also newly discovered, prompt for user choices (keep config or replace it with autodiscovered config)
3. for devices already configured but not discovered during the pairing round, prompt for user choices (keep config or delete config)
4. then set `discovery = disable`

I can see how `discovery = enable` could be useful to some road warriors who fire up their laptops in different offices with different scanners every day, but the majority of users do not change their scanners/locations that frequently and when they do, they commute between a few fixed locations.  If no UI asking to keep/replace config is possible, just do step 1 and 4 above and document in the manpage to set `discovery = pair` when a new scanner is added to the network or the existing scanners are unrecognized.

alexpevzner, 2024-02-26:
Hi @yuv,

please notice that SANE backend neither have its own user interface to prompt user nor its own persistent storage.

You may disable automatic discovery and add all devices manually, with some help from `airscan-discover` utility,

But implementing whatever you propose is very hard.

Please notice also, I'm doing my best to make discovery as fast as possible, and depending on your hardvare it may be as fast as 1 second or so.

ammgws, 2024-02-27:
I changed my config in `/etc/sane/airscan.conf` to 
```ini
[devices]
"Brother MFC-L3780CDW series" = http://10.1.1.15:80/WebServices/ScannerService, WSD

[options]
discovery = disable
```

and now `scanimage -L` takes 3.4 seconds instead of 8.4 seconds.

Then I copied `/etc/sane.d/dll.d/airscan` to `~/.config/sane/dll.conf`, and also copied `/etc/sane/airscan.conf` from above to `~/.config/sane`. Running `SANE_CONFIG_DIR=~/.config/sane/ scanimage -L`, now it takes only 10ms. Scanning starts straight away.