alexpevzner/sane-airscan #334: Ricoh Aficio MP 201 ADF 1 page scan

agapovmax, 2024-04-02:
Всем добра!

Сабж ни в какую не хочет сканировать с АПД более одной страницы. Проверял на последней версии сборки sane-airscan.

scanimage -A
```
Output format is not set, using pnm as a default.

All options specific to device `airscan:w0:RICOH Aficio MP 201':
  Standard:
    --resolution 100|200|300|400|600dpi [300]
        Sets the resolution of the scanned image.
    --mode Color|Gray [Color]
        Selects the scan mode (e.g., lineart, monochrome, or color).
    --source Flatbed|ADF|ADF Duplex [Flatbed]
        Selects the scan source (such as a document-feeder).
  Geometry:
    -l 0..216.002mm [0]
        Top-left x position of scan area.
    -t 0..356.006mm [0]
        Top-left y position of scan area.
    -x 0..216.002mm [216.002]
        Width of scan-area.
    -y 0..356.006mm [356.006]
        Height of scan-area.
  Enhancement:
    --brightness -100..100% (in steps of 1) [0]
        Controls the brightness of the acquired image.
    --contrast -100..100% (in steps of 1) [0]
        Controls the contrast of the acquired image.
    --shadow 0..100% (in steps of 1) [0]
        Selects what radiance level should be considered "black".
    --highlight 0..100% (in steps of 1) [100]
        Selects what radiance level should be considered "white".
    --analog-gamma 0.0999908..4 [1]
        Analog gamma-correction
    --negative[=(yes|no)] [no]
        Swap black and white
    --adf-justification-x <string> [inactive]
        ADF width justification (left/right/center)
    --adf-justification-y <string> [inactive]
        ADF height justification (top/bottom/center)
```

scanimage -d 'airscan:w0:RICOH Aficio MP 201' -x 216 -y 297 --source=ADF --format=jpeg --batch='scan_out3.jpeg'
```
API: sane_init(): called
loading configuration file /etc/sane.d/airscan.conf
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 10.0.0.10@eth0, ifindex=2
WSDD:  + fe80::5246:5dff:fea1:955a@eth0, ifindex=2
WSDD: 10.0.0.10: started discovery, UDP port=35909
WSDD: fe80::5246:5dff:fea1:955a: started discovery, UDP port=43217
MDNS: browse-ipv4@eth0("_ipp._tcp"): AVAHI_BROWSER_NEW (CACHED MCAST)
MDNS: browse-ipv4
[scanimage-RICOH-Aficio-MP-201.log](https://github.com/alexpevzner/sane-airscan/files/14833719/scanimage-RICOH-Aficio-MP-201.log)
@eth0("_ipp._tcp"): "RICOH Aficio MP 201 [5123E0]" (CACHED MCAST)
API: sane_open("w0:RICOH Aficio MP 201"): called
MDNS: browse-ipv6@*("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6@*("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6@*("_uscan._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6@*("_uscans._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6@*("_scanner._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
MDNS: resolve-ipv4@eth0("_ipp._tcp", "RICOH Aficio MP 201 [5123E0]"): 10.0.0.20:631 (CACHED MCAST)
WSDD: directed probe: trying if=eth0, addr=10.0.0.20
WSDD: HTTP POST http://10.0.0.20/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
WSDD: HTTP resolving 10.0.0.20 80
[... 231 lines cut ...]
RICOH Aficio MP 201: PROTO_OP_CHECK: fault code: ClientErrorJobIdNotFound
RICOH Aficio MP 201: PROTO_OP_CHECK: ScannerState: Idle
RICOH Aficio MP 201: PROTO_OP_CHECK: ScannerStateReason: None
RICOH Aficio MP 201: PROTO_OP_CHECK: decoded: status="Error during device I/O" next=PROTO_OP_FINISH delay=0
RICOH Aficio MP 201: JOB status=Error during device I/O
RICOH Aficio MP 201: DEVICE_STM_SCANNING->DEVICE_STM_DONE
RICOH Aficio MP 201: DEVICE_STM_DONE->DEVICE_STM_IDLE
RICOH Aficio MP 201: API: sane_start(): Error during device I/O
scanimage: sane_start: Error during device I/O
Batch terminated, 1 page scanned
RICOH Aficio MP 201: API: sane_close(): called
RICOH Aficio MP 201: DEVICE_STM_IDLE->DEVICE_STM_CLOSED
RICOH Aficio MP 201: removed from device table
RICOH Aficio MP 201: closed protocol "WSD"
RICOH Aficio MP 201: device destroyed
RICOH Aficio MP 201: API: sane_close(): done
API: sane_exit(): called
zeroconf: device gone urn:uuid:9bed92fe-1dd1-11b2-a906-a5c921a30cbb
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 2 (eth0)
API: sane_exit(): OK
```
Любопытно, что даже с указанием scanimage использовать формат tiff или png, в отладочной информации все равно проскакивает сохранение в JPEG. Между тем, конечный файл сохраняется корректно в указанном в параметрах формате.

Правильно ли я понимаю, что это является особенностью самого аппарата?

alexpevzner, 2024-04-02:
Добрый вечер!

> Сабж ни в какую не хочет сканировать с АПД более одной страницы

Мне бы трассу протокола. Это надо в `/etc/sane.d/airscan.conf` раскомментировать строчки:
```
[debug]
trace   = ~/airscan/trace
```
И посканировать разок. Директория `~/airscan/trace` создастся автоматически и в ней появятся файлы `*.log` и `*.tar`. Мне бы пока только `*.log`, тары не нужны.

В процессе разбирательств Вы сможете сами пересобрать airscan?

> Правильно ли я понимаю, что это является особенностью самого аппарата?

Драйвер SANE всегда отдает приложению (в данном случае это `scanimage`) распакованное изображение. И `scanimage` сам его уже перепаковывает в то, что попросили.

JPEG, который проскакивает в логе - это формат, в котором изображение передается между драйвером и устройством. Устройство предлагает некоторый набор форматов, драйвер из них выбирает на свое усмотрение.

Большинство устройст поддерживает JPEG, но бывают и исключения.

agapovmax, 2024-04-02:
> В процессе разбирательств Вы сможете сами пересобрать airscan?

Да, разумеется.

Файл трассировки приложил.
[scanimage-RICOH-Aficio-MP-201.log](https://github.com/alexpevzner/sane-airscan/files/14842458/scanimage-RICOH-Aficio-MP-201.log)

alexpevzner, 2024-04-03:
Как-то не очень понятно. По логу я вижу, что пришло две картинки. При этом сканирование было одностороннее. Т.е., два листа с податчика сканер сгреб, отсканировал и прислал. А сколько там всего было, я не знаю.

Кстати, если у вас выходной формат JPEG, он не поддерживает многостраничные документы. Многостраничные документы поддерживает TIFF (но я не уверен, что scanimage умеет этим пользоваться) и PDF.

alexpevzner, 2024-04-03:
А следующий скан нормально забирает следущую страницу, или есть какие-то пропуски?

agapovmax, 2024-04-04:
>А следующий скан нормально забирает следущую страницу, или есть какие-то пропуски?

Пропусков нет, следующая попытка сканирования с АПД берет одну страницу.

Да, `simple-scan` действительно показывает две страницы (одна с текстом, другая пустая), в то время, как за раз сканирует только один лист.

Сканирование с АДП с установленными листами. `simple-scan` с настройками "Текст"+"Все страницы с автоподатчика"  [simple-scan-RICOH-Aficio-MP-201_all_pages_once.log](https://github.com/alexpevzner/sane-airscan/files/14862613/simple-scan-RICOH-Aficio-MP-201_all_pages_once.log)
Сканирование с настройками "Текст"+"Одна страница" выполненное дважды [simple-scan-RICOH-Aficio-MP-201_one_page_twice.log](https://github.com/alexpevzner/sane-airscan/files/14862615/simple-scan-RICOH-Aficio-MP-201_one_page_twice.log)
Сканирование с пустым АПД с настройками "Текст"+"Все страницы с автоподатчика" [simple-scan-RICOH-Aficio-MP-201_empty.log](https://github.com/alexpevzner/sane-airscan/files/14862616/simple-scan-RICOH-Aficio-MP-201_empty.log)

alexpevzner, 2024-04-04:
Надо же, ему целых 10 секунд надо, чтобы с пустым автоподатчиком сообразить, что кина не будет.

Еще одна просьба. Пересоберите, пожалуйста, драйвер из свежих исходников и запустите програмку `test`, которая там заодно соберется. Она сделает одну попытку скана и тоже породит трассу `test-...log`, пришлите мне эту трассу тоже.

Будет попытка скана удачной или нет, не важно. Я там просто запросил дополнительную информацию о сканере и хочу на нее посмотреть.

agapovmax, 2024-04-05:
Прикладываю 
[test-RICOH-Aficio-MP-201.log](https://github.com/alexpevzner/sane-airscan/files/14881079/test-RICOH-Aficio-MP-201.log)

alexpevzner, 2024-04-05:
Кажется, это не самый последний коммит, b7fd125514676060247fe69cb7ca540ff150238c туда почему-то не попал...

agapovmax, 2024-04-05:
Извиняюсь, сделал `git pull`, увидел коммит для airscan-wsd.c, пересобрал 
[test-RICOH-Aficio-MP-201.log](https://github.com/alexpevzner/sane-airscan/files/14881420/test-RICOH-Aficio-MP-201.log)

alexpevzner, 2024-04-05:
А попробуйте, пожалуйста, в файле `airscan-wsd.c` в строке 807 найти такую строчку:
```
    xml_wr_add_text(xml, "sca:ImagesToTransfer", "0");
```
И "0" заменить на "10", а потом пересобрать и посканировать с податчика. Только уже не тестом, а настоящей программой.

Я обычно не говорю при таких экспериментах make install, а просто делаю символическую ссылку: `/lib64/sane/libsane-airscan.so.1 -> /home/pzz/c/pzz/sane-airscan/libsane-airscan.so.1`

У меня sane-вские драйвера находятся в `/lib64/sane/...`, но в некоторых дистрибутивах они находятся в `/lib/sane/...`. А `/home/pzz/c/pzz/sane-airscan/` - это билдовая директория, где я работаю (подставьте вместо нее свою).

Этого достаточно, чтобы использовался свежесобраный драйвер, и не надо после каждой пересборки чего-либо переставлять.

agapovmax, 2024-04-05:
Да! Аппарат взял три тестовые страницы с АПД и успешно их просканировал! Файлы создаются.
Но, сканирование завершается ошибкой фронтенда "Error during device I/O"
[scanimage-RICOH-Aficio-MP-201.log](https://github.com/alexpevzner/sane-airscan/files/14883362/scanimage-RICOH-Aficio-MP-201.log)

alexpevzner, 2024-04-05:
А попробуйте какое-нибудь большое число. Например, "1000".

Это бага в прошивке. "0" - значит сканировать, пока листочки не кончатся. Но сканер это воспринимает, как "1".

Беда в том, что не вижу надежного способа отличить эту модель от каких-то других. И боюсь, среди других найдутся такие, которые не будут нормально работать с большим числом. А их мне ломать тоже не хочется :(

> Но, сканирование завершается ошибкой фронтенда "Error during device I/O"

Это как раз легко лечится. Этот сканер сообщает о том, что листочки кончились, нестандартным способом. А я все непонятные ошибки возвращаю, как "I/O error". Просто добавлю эту в список понятных.

agapovmax, 2024-04-05:
Параметр `1024` отработал без ошибок.

Оставлю данные о версии FW устройства
```
Система: 1.10
NIB: 9.08
Web Image Monitor: 1.02
```

Похоже список совместимых устройств можно дополнить сабжем!

alexpevzner, 2024-04-06:
> Но, сканирование завершается ошибкой фронтенда "Error during device I/O"

Это я, надеюсь, починил. Попробуйте, пожалуйста, в двух вариантах:
1. В податчике что-то есть. Должно отсканироваться все, что есть, а потом "ADF empty" (которая, вероятно, не будет воспринята, как ошибка - это ожидаемое завершение сканирования с податчика)
2. Податчик изначально пустой - должно быть правильное сообщение об ошибке

Насчет поставить лимит "1024" вместо "0" я еще подумаю. Боюсь другое какое-нибудь капризное устройство сломать, а как понять, что я общаюсь именно в этим (или, хотя бы, именно с каким-то Ricoh), пока не понимаю: в протоколе WSD, к сожалению, такого удобства не предусмотрено.

agapovmax, 2024-04-07:
> Боюсь другое какое-нибудь капризное устройство сломать, а как понять, что я общаюсь именно в этим (или, хотя бы, именно с каким-то Ricoh), пока не понимаю: в протоколе WSD, к сожалению, такого удобства не предусмотрено.

А возможно использовать модель из вывода zeroconf?
```
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 2 (eth0)
zeroconf:   name:      -
zeroconf:   model:     RICOH Aficio MP 201
zeroconf:   addresses:
zeroconf:     10.0.0.20
zeroconf:   protocol:  WSD
zeroconf:   endpoints:
zeroconf:     http://10.0.0.20:53002
zeroconf:   device:    0000 (created)
zeroconf: device_list wait: WSDD not finished...

```

Например, для Pantum 5100, модель (model) - `Pantum BM5100ADW Series` не редактируемое значение. 
```
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
zeroconf: found urn:uuid:1dfd422b-4dd9-400e-8047-47282ad2c0da
zeroconf:   method:    ZEROCONF_USCANS_TCP
zeroconf:   interface: 2 (eth0)
zeroconf:   name:      Pantum BM5100ADW Series 4B94CA
zeroconf:   model:     Pantum BM5100ADW Series
zeroconf:   addresses:
zeroconf:     10.0.0.23
zeroconf:   protocol:  eSCL
zeroconf:   endpoints:
zeroconf:     https://10.0.0.23:443/eSCL/
zeroconf:   device:    0000 (found)
zeroconf: device_list wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: Pantum BM5100ADW Series 4B94CA (0): can: eSCL, use: eSCL
zeroconf: zeroconf_device_list_get: resulting list:
zeroconf:   eSCL  "e0:Pantum BM5100ADW Series 4B94CA"
```
В отличии от Служба Bonjour (name) - `Pantum BM5100ADW Series 4B94CA`
[image]

alexpevzner, 2024-04-07:
> А возможно использовать модель из вывода zeroconf?

Ну я же работаю не только с устройствами, которых нашел zeroconf, но и можно просто в конфиге URL задать.

Я пока колеблюсь между тем, чтобы рискнуть попробовать задать большое число с запасом, в надежде, что это никого не сломает, или запросить имя модели пol IPP (принтеровский протокол). Но по IPP довольно муторно это делать...

agapovmax, 2024-04-08:
> В податчике что-то есть. Должно отсканироваться все, что есть, а потом "ADF empty" (которая, вероятно, не будет воспринята, как ошибка - это ожидаемое завершение сканирования с податчика)

Сканирование заканчивается штатно. Ошибок и информационных сообщений из фронтенда не выходит.
> Податчик изначально пустой - должно быть правильное сообщение об ошибке

Ошибка "Error during device I/O"

Дополнительное тестирование выявило несколько особенностей! 
1. АПД в двухстороннем режиме. Второй лист сканируется в перевернутом виде.
2. АПД. После сканирования каждого листа происходит задержка - переключается лампа сканера с АПД на планшет и обратно. 

https://github.com/alexpevzner/sane-airscan/assets/16970072/3a8b6504-caf1-4f54-8c0a-6ad10da3bd3e

alexpevzner, 2024-04-08:
> АПД в двухстороннем режиме. Второй лист сканируется в перевернутом виде.

А можно мне трассу с `.tar`-ом от этого случая? И не кладите важных листочков в сканер; `.tar` содержит картинки.

> После сканирования каждого листа происходит задержка - переключается лампа сканера с АПД на планшет и обратно.

А там нижнюю сторону автоподатчика не обслуживает ли сенсор планшета?

agapovmax, 2024-04-08:
Прикладываю .zip и .log сканирования двух страниц. Каждый лист и его сторона подписаны
[scanimage-RICOH-Aficio-MP-201.log](https://github.com/alexpevzner/sane-airscan/files/14905555/scanimage-RICOH-Aficio-MP-201.log)
[scanimage-RICOH-Aficio-MP-201.zip](https://github.com/alexpevzner/sane-airscan/files/14905578/scanimage-RICOH-Aficio-MP-201.zip)

> А там нижнюю сторону автоподатчика не обслуживает ли сенсор планшета?

Нет, сенсор планшета даже не используется.
Лист заходит в АПД, сканируется через лампу АПД, затем после прогона первой стороны, загорается лампа планшета, гаснет, затем загорается лампа АПД и вторая сторона сканируется.

alexpevzner, 2024-04-08:
Т.е., он делает двухстороннее сканирование в два прохода? Тогда понятно, почему картинка перевернута - задняя сторона проходит через сенсор задом наперед.

Мда. Надо как-то научиться надежно опеделять этот девайс...

alexpevzner, 2024-06-01:
@agapovmax, добрый вечер!

Я попытался "автоматизировать" тот workaround для вашего сканера, который я тогда просил Вас сделать руками. Не могли бы вы попробовать `sane-airscan` на Вашем устройстве и подтвердить, что всё ОК?

closed 2024-11-18 by alexpevzner