alexpevzner/sane-airscan #335: HP LaserJet 500 color MFP M575 using eSCL: Scanner operates, histogram data shown in xsane, but error: "Failed to start scanner: Document feeder out of documents"

rubin55, 2024-04-13:
I added my MFP device, an HP LaserJet 500 color MFP M575 which is connected through ethernet and has a fixed IP address, to `/etc/sane.d/escl.conf`:

```
# escl.conf -- ESCL configuration
device https://172.17.1.26:443 "HP LaserJet 500 color MFP M575"
```

I then start `xsane`, select the scanner from the list and click the `scan` button. I can hear + see the scanner motor operating. Looking at the device status page, I see a message "Scanning page from glass". I see `xsane` loading the color histogram data, but I then get an error message dialog telling me the following:

```
Failed to start scanner: Document feeder out of documents
```

It does not matter if I select `flatbed` or `ADC` or `ADC Duplex` as source. It also does not matter if I do scan to file directly, use preview or send to viewer, I always get the same error message.

Please find attached the trace log:
[xsane-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/14967408/xsane-zeroconf.log)

alexpevzner, 2024-04-14:
Hi @rubin55,

there are actually 2 eSCL drivers for SANE: `sane-airscan` and `sane-escl`.

If you've added your device using `escl.conf`, you use `sane-escl`.

I suggest you to try my driver. Usually it works better. You will need to add your device to the `/etc/sane.d/airscan.conf` using the following syntax:

```
[devices]
"HP LaserJet 500 color MFP M575" = https://172.17.1.26:443/eSCL
```
Or you can use the `airscan-discover` program (typically it comes with the `sane-airscan` package), it will discover all devices around and create `airscan.conf` file for you.

Please notice, you may use arbitrary device name (quoted, left to the equal sign); this is the name that you will see in device selection menu of the scanning program

rubin55, 2024-04-19:
It works! Thanks so much, I can't tell you how long I was struggling with this, such an improvement to my workflow. I did notice that in xsane, I can only select greyscale or color, not black and white; is that a limitation of the protocol of sorts? In any case, I'm super happy, thanks a bunch.

rubin55, 2024-04-24:
By the way, is it expected that `xsane` only shows a `Color` and `Gray` scan mode? I would expect `B/W` also, but for some reason, it doesn't show it.

alexpevzner, 2024-04-24:
sane-airscan doesn't support B&W. To support B&W I will have to add PDF decoder. It adds a bit too much dependencies, plus PDF hard to decode line-by-line, without need to allocate a large whole-image buffer.

So taking in account low value of B&W scan (it easily can be converted from the color or grayscale image without need to implement it in the driver) currently I think it doesn't worse to be implemented.

However there are always some chance that I may change my mind in the future :)

closed 2024-05-30 by alexpevzner