alexpevzner/sane-airscan #337: Key at Debian Repo has expired

sysms, 2024-04-24:
Dear maintainer,

the key at the Debian Repository has expired. 

`Err:10 http://download.opensuse.org/repositories/home:/pzz/Debian_10  InRelease
  The following signatures were invalid: EXPKEYSIG 23E07629B9E3C9E4 home:pzz OBS Project <home:pzz@build.opensuse.org>
`

Please renew it. :) 

Thanks!

sysms, 2024-04-26:
@alexpevzner  you're right of course :D  

it's only one old internal system which can't be updated currently for reasons, which still has this external repository included.
that's why it has popped up into my eyes.

so no worries - I just wanted to let you know, just in case someone else also still uses this packages.

Thanks for your support in building these packages!

MR-bgg-c, 2024-07-09:
Hello. The sane-airscan instructions "[Add repository and install manually](https://software.opensuse.org//download.html?project=home%3Apzz&package=sane-airscan#manualUbuntu)" for xUbuntu 20.04 fail on `sudo apt update` with:
```
Get:9 http://download.opensuse.org/repositories/home:/pzz/xUbuntu_20.04  InRelease [1,514 B]
Err:9 http://download.opensuse.org/repositories/home:/pzz/xUbuntu_20.04  InRelease
  The following signatures were invalid: EXPKEYSIG 23E07629B9E3C9E4 home:pzz OBS Project <home:pzz@build.opensuse.org>
Reading package lists... Done
W: GPG error: http://download.opensuse.org/repositories/home:/pzz/xUbuntu_20.04  InRelease: The following signatures were invalid: EXPKEYSIG 23E07629B9E3C9E4 home:pzz OBS Project <home:pzz@build.opensuse.org>
E: The repository 'http://download.opensuse.org/repositories/home:/pzz/xUbuntu_20.04  InRelease' is not signed.
```

I assume the xUbuntu 20.04 installation instructions should work on any variety of Ubuntu 20.04, not only on xUbuntu 20.04 .

alexpevzner, 2024-07-22:
Hi, I believe, repositories now work. At least those that I've tried. Please confirm and sorry for delay

MR-bgg-c, 2024-07-23:
Thanks the sane-airscan instructions "[Add repository and install manually](https://software.opensuse.org//download.html?project=home%3Apzz&package=sane-airscan#manualUbuntu)" for xUbuntu 20.04 work now.

alexpevzner, 2024-07-24:
Sorry that it took so long time...

closed 2024-07-24 by alexpevzner

MR-bgg-c, 2024-07-24:
Thanks for getting to it - and for maintaining the project :).
Maybe going forward include in the update/release cycle a test
installation according to the instructions, saves some headaches.