alexpevzner/sane-airscan #338: airscan cannot discover WSD protocol but WSDD can find it on Xerox Workcentre 3025

demanuPL, 2024-05-12:
I'm trying to discover my Xerox WorkCentre 3025 but my OS (Opensuse Thumbleweed) cannot detect supported protocol for it which in this case should be WSD, but on logs I can see that WSDD is succesfully connecting to it. 

correct address of my Xerox device is 10.88.88.204

Here is my complete debug log from scanimage -L

```scanimage -L
API: sane_init(): called
loading configuration file /etc/sane.d/airscan.conf
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + fdf2:3d78:39b8:0:5554:6835:88ba:b980@wlp4s0, ifindex=3
WSDD:  + fdf2:3d78:39b8:0:e23c:8aec:40fa:1a75@wlp4s0, ifindex=3
WSDD:  + 10.88.88.202@wlp4s0, ifindex=3
WSDD:  + fe80::f4f4:6cc6:db15:8214@wlp4s0, ifindex=3
WSDD:  + fd7a:115c:a1e0::6435:aa40@tailscale0, ifindex=5
WSDD:  + 100.117.170.64@tailscale0, ifindex=5
WSDD:  + fe80::421e:a085:6340:5430@tailscale0, ifindex=5
WSDD:  + 172.17.0.1@docker0, ifindex=6
WSDD: fdf2:3d78:39b8:0:5554:6835:88ba:b980: started discovery, UDP port=33423
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: fdf2:3d78:39b8:0:e23c:8aec:40fa:1a75: started discovery, UDP port=36880
WSDD: 10.88.88.202: started discovery, UDP port=58143
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: fe80::f4f4:6cc6:db15:8214: started discovery, UDP port=57605
WSDD: fd7a:115c:a1e0::6435:aa40: started discovery, UDP port=41460
WSDD: 100.117.170.64: started discovery, UDP port=55839
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: fe80::421e:a085:6340:5430: started discovery, UDP port=56038
WSDD: 172.17.0.1: started discovery, UDP port=37686
API: sane_get_devices(): called
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv4@wlp4s0("_ipp._tcp"): AVAHI_BROWSER_NEW (CACHED MCAST)
MDNS: browse-ipv4@wlp4s0("_ipp._tcp"): "Xerox WorkCentre 3025 (XRX9C934EC0C407)" (CACHED MCAST)
MDNS: browse-ipv6@*("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6@*("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6@*("_uscan._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6@*("_uscans._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv4@wlp4s0("_scanner._tcp"): AVAHI_BROWSER_NEW (CACHED MCAST)
MDNS: browse-ipv4@wlp4s0("_scanner._tcp"): "Xerox WorkCentre 3025 (XRX9C934EC0C407)" (CACHED MCAST)
MDNS: browse-ipv6@*("_scanner._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: resolve-ipv4@wlp4s0("_ipp._tcp", "Xerox WorkCentre 3025 (XRX9C934EC0C407)"): 10.88.88.204:631 (CACHED MCAST)
WSDD: directed probe: trying if=wlp4s0, addr=10.88.88.204
WSDD: HTTP POST http://10.88.88.204/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
WSDD: HTTP resolving 10.88.88.204 80
WSDD: HTTP trying 10.88.88.204:80
WSDD: HTTP fdpoll: {--} -> {-w}
MDNS: resolve-ipv4@wlp4s0("_scanner._tcp", "Xerox WorkCentre 3025 (XRX9C934EC0C407)"): 10.88.88.204:9400 (CACHED MCAST)
WSDD: directed probe: trying if=wlp4s0, addr=10.88.88.204
WSDD: directed probe: already in progress, skipping
WSDD: HTTP 803 bytes sent
WSDD: HTTP done request sending
WSDD: HTTP fdpoll: {-w} -> {rw}
WSDD: HTTP fdpoll: {rw} -> {r-}
WSDD: HTTP 1500 bytes received
WSDD: HTTP POST http://10.88.88.204/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: got response headers (200)
WSDD: HTTP 23 bytes received
WSDD: HTTP done response reception
WSDD: HTTP POST http://10.88.88.204/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: 200 OK
WSDD: HTTP POST http://10.88.88.204/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: OK
WSDD: HTTP POST http://10.88.88.204:8018/wsd
WSDD: HTTP POST http://192.168.3.1:8018/wsd
WSDD: HTTP resolving 10.88.88.204 8018
WSDD: HTTP trying 10.88.88.204:8018
WSDD: HTTP fdpoll: {--} -> {-w}
WSDD: HTTP resolving 192.168.3.1 8018
WSDD: HTTP trying 192.168.3.1:8018
WSDD: HTTP fdpoll: {--} -> {-w}
WSDD: HTTP 673 bytes sent
WSDD: HTTP done request sending
WSDD: HTTP fdpoll: {-w} -> {rw}
WSDD: HTTP fdpoll: {rw} -> {r-}
WSDD: HTTP 1460 bytes received
WSDD: HTTP POST http://10.88.88.204:8018/wsd: got response headers (200)
WSDD: HTTP 1460 bytes received
WSDD: HTTP 488 bytes received
WSDD: HTTP done response reception
WSDD: HTTP POST http://10.88.88.204:8018/wsd: 200 OK
WSDD: HTTP POST http://10.88.88.204:8018/wsd: OK
MDNS: browse-ipv6@*("_ipp._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6@*("_ipps._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6@*("_uscan._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6@*("_uscans._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6@*("_scanner._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: "Xerox WorkCentre 3025 (XRX9C934EC0C407)": publish-delay timer expired
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: found urn:uuid:16a65700-007c-1000-bb49-9c934ec0c407
zeroconf:   method:    ZEROCONF_MDNS_HINT
zeroconf:   interface: 3 (wlp4s0)
zeroconf:   name:      Xerox WorkCentre 3025 (XRX9C934EC0C407)
zeroconf:   model:     Xerox WorkCentre 3025
zeroconf:   addresses:
zeroconf:     10.88.88.204
zeroconf:   device:    0000 (created)
zeroconf: device_list wait: WSDD not finished...
WSDD: 172.17.0.1: done discovery
WSDD: fdf2:3d78:39b8:0:5554:6835:88ba:b980: done discovery
WSDD: 10.88.88.202: done discovery
WSDD: fe80::421e:a085:6340:5430: done discovery
WSDD: fdf2:3d78:39b8:0:e23c:8aec:40fa:1a75: done discovery
WSDD: fe80::f4f4:6cc6:db15:8214: done discovery
WSDD: fd7a:115c:a1e0::6435:aa40: done discovery
WSDD: 100.117.170.64: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: device_list wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: Xerox WorkCentre 3025 (XRX9C934EC0C407) (0): can: none, use: none
zeroconf: Xerox WorkCentre 3025 (XRX9C934EC0C407) (0): skipping, none of supported protocols discovered
zeroconf: zeroconf_device_list_get: resulting list:
API: sane_get_devices(): done

No scanners were identified. If you were expecting something different,
check that the scanner is plugged in, turned on and detected by the
sane-find-scanner tool (if appropriate). Please read the documentation
which came with this software (README, FAQ, manpages).
API: sane_exit(): called
WSDD: HTTP POST http://192.168.3.1:8018/wsd: Cancelled
zeroconf: device gone urn:uuid:16a65700-007c-1000-bb49-9c934ec0c407
zeroconf:   method:    ZEROCONF_MDNS_HINT
zeroconf:   interface: 3 (wlp4s0)
API: sane_exit(): OK
```

A you can see on bottom there is strage error:
`zeroconf: Xerox WorkCentre 3025 (XRX9C934EC0C407) (0): skipping, none of supported protocols discovered`

alexpevzner, 2024-05-15:
Hi @demanuPL,

thank you for your report. What version of `sane-airscan` do you use?

alexpevzner, 2024-05-30:
hi @demanuPL,

I believe I've already fixed a similar problem a while ago. Could you please let me know what `sane-airscan` version do you use?

closed 2024-11-18 by alexpevzner