alexpevzner/sane-airscan #339: Epson Workforce WF-3520 cannot scan duplex via airscan

Signum, 2024-05-21:
My WF-3520 only scans the upper side of every page fed into the ADF when using airscan.

The device was shown as "airscan:w1:EPSON WF-3520 Series".

After removing sane-airscan, SANE fell back to the "epson2" driver that indeed managed to scan duplex. It is shown as device name "epson2:net:printer" ("printer" being the name on my network).

I cannot tell if that's a shortcoming of the Airscan protocol on the Epson printers (which did not receive a firmware update in years) or of sane-airscan. If it is unfixable then this report may at least help others who kept banging their head against the wall. Dummy solution for Debian/Ubuntu/Mint users out there: "apt purge sane-airscan" until this is fixed.

alexpevzner, 2024-05-23:
Hi @Signum,

I need to see the protocol trace. To enable it, please uncomment the following lines in the `/etc/sane.d/airscan.conf` file:
```
[debug]
trace = ~/airscan/trace
```
Directory `~/airscan/trace` will be created automatically and trace files will be written there.

For now I only need the `*.log` files, `*.tar` files are not needed.

Signum, 2024-05-29:
FYI: I had the "sane-airscan" package (Linux Mint, 0.99.27-1build1) removed for the last days and just reinstalled it.

FYI: I have bought a new printer (and kept the old one) Epson Workforce WF-3840.

Test 1: Epson Workforce WF-3520.
device `airscan:w2:EPSON WF-3520 Series' is a WSD EPSON WF-3520 Series ip=192.168.15.112
Result: Only front side is scanned.
Trace: 
[wf-4830.zip](https://github.com/alexpevzner/sane-airscan/files/15482525/wf-4830.zip)
(I had to ZIP them because Github did not like .tar or .log files.)

Test 2: Epsn Workforce WF-4830.
device `airscan:e0:EPSON WF-4830 Series' is a eSCL EPSON WF-4830 Series ip=192.168.15.200
Result: Both sides of the page are scanned. However the second page is upside down (aka: rotated 180 degrees)
Trace: 
[wf-3520.zip](https://github.com/alexpevzner/sane-airscan/files/15482534/wf-3520.zip)

alexpevzner, 2024-05-30:
Hi @Signum,

I believe I've fixed duplex scanning on Epson Workforce WF-3520. I will appreciate if you could build `sane-airscan` from the Github master and retest.

There is actually no need to `make install` after build - it is enough to create a symlink from `/usr/lib64/sane/libsane-airscan.so.1` (or `/usr/lib/..., depending on your distro) into your build directory.

Regarding Epson Workforce WF-4830, well, I'm not sure. There are yet another device, `RICOH Aficio MP-201` with the similar problem. Details are under #334, but discussion there is in Russian, so may be hard to read, sorry :)

I guess both devices have physically a single-side ADF scanner, and scan both sides by doing a second pass after physically turning over a sheet of paper being scanning, correct? And during the second pass, a back side traverses the scanner in the reverted position.

In theory this is a protocol violation and devices should not behave this way. But probably there is not enough hardware resources in the scanner (memory and so on) to do things correctly, so they do their best...

This issue requires some thinking. Reverting JPEG image in software (on a host) is possible, but it is noticeable amount of work.

Signum, 2024-05-31:
@alexpevzner Thank you very much. I hope you realized that I mixed up the ZIP files in my previous comment. But the files were named after the respective printer so it probably made sense anyway.

I built the master tree and replaced the libsane-airscan.so.1.

WF-3520 now scans duplex, but with the even pages rotated 180 degrees:
[image]

And, yes, both the WF-33520 and the WF-4830 mechanically turn the page around when scanning the back side.

I also tend to believe that it's a protocol violation. The "Epson iPrint" Android app explicitly asks whether to rotate even pages. Technically we can blame Epson. From a user perspective it may look like a SANE driver problem. :)

closed 2024-11-18 by alexpevzner