alexpevzner/sane-airscan #340: Release version of sane-airscan not working for me on openSUSE Tumbleweed

dhwalker, 2024-05-23:
Sane-airscan stopped working for me recently on recent Tumbleweed snapshots. On Tumbleweed 20240512, aarch64, xsane displays a "No devices available" message and does not start.  Airscan-discover does find it however:

```
    > airscan-discover
     [devices]
       Canon MF650C Series = [http://192.168.1.130:80/eSCL/](http://192.168.1.130/eSCL/), eSCL
       Canon MF650C Series = http://192.168.1.130/active/msu/scan, WSD

    > sudo zypper se --details --installed-only sane
    Loading repository data...
    Reading installed packages...

    S  | Name                     | Type    | Version     | Arch    | Repository
    ---+--------------------------+---------+-------------+---------+------------------------
    i  | libsane1                 | package | 1.3.0-1.1   | aarch64 | openSUSE-Tumbleweed-Oss
    i+ | sane-airscan             | package | 0.99.29-1.1 | aarch64 | openSUSE-Tumbleweed-Oss
    i  | sane-backends            | package | 1.3.0-1.1   | aarch64 | openSUSE-Tumbleweed-Oss
    i  | sane-backends-autoconfig | package | 1.3.0-1.1   | noarch  | openSUSE-Tumbleweed-Oss
    i+ | xsane                    | package | 0.999-4.3   | aarch64 | openSUSE-Tumbleweed-Oss
```

After some discussion on https://bugzilla.suse.com/show_bug.cgi?id=1221512 (" Bug 1221512 - scanner MFC-J5620DW not anymore recognized by xsane or YaST2"), I found that the sane-airscan-unstable package in home:pzz works for me. So, I have two requests:

- Would you update the version of sane-airscan in openSUSE-Tumbleweed-Oss?
- I usually do my scanning from a Raspberry Pi 4, so would you have OBS start building sane-airscan for aarch64?

Thanks, and particularly thanks for sane-airscan.

David Walker

alexpevzner, 2024-05-24:
Hi @dhwalker,

May I see the protocol trace? To enable it, please uncomment the following lines in the `/etc/sane.d/airscan.conf` file:
```
[debug]
trace = ~/airscan/trace
```
Directory `~/airscan/trace` will be created automatically and trace files will be written there.

For now I only need the `*.log` files, `*.tar` files are not needed.

dhwalker, 2024-05-25:
Thanks for responding, @alexpevzner.

The ~/airscan/trace directory is not created (and no *.log or *.tar files are created) when using sane-airscan. I am attaching the *.log files for sane-airscan-unstable.

David

[xsane-Canon-MF650C-Series.log](https://github.com/alexpevzner/sane-airscan/files/15441008/xsane-Canon-MF650C-Series.log)
[xsane-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/15441009/xsane-zeroconf.log)

alexpevzner, 2024-05-25:
Looks that `sane-airscan` doesn't load at all. Could you please run `ldd /usr/lib64/sane/libsane-airscan.so.1` and show me output?

Probably some libraries that `sane-airscan` depends on is missed. I bet it is `libtiff`, I've recently added dependency on it.

If installing libtiff RPM helps, please let me know and report it to SUSE too, they should update their RPM .spec file.

dhwalker, 2024-05-25:
Interesting. The `sane-airscan` package doesn't even have `/usr/lib64/sane/libsane-airscan.so.1`. Here's some information:

```
sane-airscan installed
======================

> ldd /usr/lib64/sane/libsane-airscan.so.1 
ldd: /usr/lib64/sane/libsane-airscan.so.1: No such file or directory
> ls /usr/lib64/sane/*airscan*
ls: cannot access '/usr/lib64/sane/*airscan*': No such file or directory

> rpm -ql sane-airscan
/etc/sane.d/airscan.conf
/etc/sane.d/dll.d/airscan
/usr/bin/airscan-discover
/usr/share/doc/packages/sane-airscan
/usr/share/doc/packages/sane-airscan/README.md
/usr/share/licenses/sane-airscan
/usr/share/licenses/sane-airscan/COPYING
/usr/share/licenses/sane-airscan/LICENSE
/usr/share/man/man1/airscan-discover.1.gz
/usr/share/man/man5/sane-airscan.5.gz

sane-airscan-unstable installed
===============================

> ldd /usr/lib64/sane/libsane-airscan.so.1
	linux-vdso.so.1 (0x00007f1f971dd000)
	libavahi-common.so.3 => /lib64/libavahi-common.so.3 (0x00007f1f9714a000)
	libavahi-client.so.3 => /lib64/libavahi-client.so.3 (0x00007f1f97137000)
	libxml2.so.2 => /lib64/libxml2.so.2 (0x00007f1f96fc8000)
	libgnutls.so.30 => /lib64/libgnutls.so.30 (0x00007f1f96c00000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f1f96edf000)
	libjpeg.so.8 => /lib64/glibc-hwcaps/x86-64-v3/libjpeg.so.8.3.2 (0x00007f1f96b07000)
	libpng16.so.16 => /lib64/glibc-hwcaps/x86-64-v3/libpng16.so.16.43.0 (0x00007f1f96e92000)
	libtiff.so.6 => /lib64/libtiff.so.6 (0x00007f1f96a79000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f1f96800000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1f971df000)
	libdbus-1.so.3 => /lib64/libdbus-1.so.3 (0x00007f1f96e3d000)
	libz.so.1 => /lib64/glibc-hwcaps/x86-64-v3/libz.so.1.3.1 (0x00007f1f96e23000)
	liblzma.so.5 => /lib64/glibc-hwcaps/x86-64-v3/liblzma.so.5.4.2 (0x00007f1f96a40000)
	libjitterentropy.so.3 => /lib64/libjitterentropy.so.3 (0x00007f1f96e19000)
	libp11-kit.so.0 => /lib64/libp11-kit.so.0 (0x00007f1f96679000)
	libidn2.so.0 => /lib64/libidn2.so.0 (0x00007f1f96a1e000)
	libunistring.so.5 => /lib64/libunistring.so.5 (0x00007f1f964c4000)
	libtasn1.so.6 => /lib64/libtasn1.so.6 (0x00007f1f96a07000)
	libnettle.so.8 => /lib64/glibc-hwcaps/x86-64-v3/libnettle.so.8.8 (0x00007f1f9646c000)
	libhogweed.so.6 => /lib64/glibc-hwcaps/x86-64-v3/libhogweed.so.6.8 (0x00007f1f96421000)
	libgmp.so.10 => /lib64/libgmp.so.10 (0x00007f1f9637a000)
	libzstd.so.1 => /lib64/glibc-hwcaps/x86-64-v3/libzstd.so.1.5.6 (0x00007f1f962c9000)
	libjbig.so.2 => /lib64/libjbig.so.2 (0x00007f1f962bc000)
	libsystemd.so.0 => /lib64/libsystemd.so.0 (0x00007f1f961d8000)
	libffi.so.8 => /lib64/libffi.so.8 (0x00007f1f961cd000)
	libcap.so.2 => /lib64/libcap.so.2 (0x00007f1f961c1000)
	libgcrypt.so.20 => /lib64/glibc-hwcaps/x86-64-v3/libgcrypt.so.20.4.3 (0x00007f1f96074000)
	liblz4.so.1 => /lib64/glibc-hwcaps/x86-64-v3/liblz4.so.1.9.4 (0x00007f1f96050000)
	libgpg-error.so.0 => /lib64/libgpg-error.so.0 (0x00007f1f9602a000)

> rpm -ql sane-airscan-unstable
/etc/sane.d
/etc/sane.d/airscan.conf
/etc/sane.d/dll.d
/etc/sane.d/dll.d/airscan
/usr/bin/airscan-discover
/usr/lib64/sane
/usr/lib64/sane/libsane-airscan.so.1
/usr/share/man/man1/airscan-discover.1.gz
/usr/share/man/man5/sane-airscan.5.gz
```
However... As I was poking around, I found there's a `libsane-airscan1` package that contains (only) `/usr/lib64/sane/libsane-airscan.so.1`. I installed it with `sane-airscan`, and now xsane starts and finds my scanner. I'm not sure why `libsane-airscan1` is a separate package, but it at least it should be required by `sane-airscan`. If that sounds right to you, I'll send that fix to the Tumbleweed maintainers.

alexpevzner, 2024-05-25:
Hi @dhwalker,

yes, you are absolutely correct, the problem is in the missed dependency. Please, report it to the Tumbleweed maintainers.

My own packages brings everything (airscan-discover + sane-airscan) in one single RPM. For some strange reason, Fedora comes with two packages, like SUSE, but with correct dependency between them. SUSE seems have repeated it after Fedora, but missed dependency.

So I close the issue, as resolved and hope SUSE will eventually fix their side too.

closed 2024-05-25 by alexpevzner

dhwalker, 2024-05-25:
Done: https://bugzilla.opensuse.org/show_bug.cgi?id=1225289

Thanks for you help!