alexpevzner/sane-airscan #341: Is there any way to specify The IP address of the Deivce instead of using broadcast

ihipop, 2024-06-14:
I connected to  my office in another broadcast-isolated subnet， eg 
VPN:192.168.2.0/24
LAN:192.168.1.0/24
These two subnets can communicate with each other,   but the broadcast is isolated because they are in different subnets with different  broadcast address

the official [BROTHER iPrint&Scan](https://play.google.com/store/apps/details?id=com.brother.mfc.brprint&hl=en) APP can Add printer &scaner via IP address, so it can access printers&scaners accross the subnets

[image]

alexpevzner, 2024-06-14:
Yes, there are configuration file: `/etc/sane.d/airscan.conf`

You may manually add devices into the appropriate section:

```
[devices]
"Kyocera MFP Scanner" = http://192.168.1.102:9095/eSCL
```

There is also the `airscan-discover` utility, which will look around and generate a template of the `[devices]` section automatically. If your network is split into several broadcast domains, you may run it once in the computer, connected to the same broadcast domain as your scanner and then save and use the generated configuration.

Note, the path part of device URL is almost always **/eSCL** for the eSCL devices, but very different and hard to guess for WSD devices.

closed 2024-06-14 by alexpevzner

ihipop, 2024-06-14:
Thanks for your kindly and quick response

>Note, the path part of device URL is almost always /eSCL for the eSCL devices, but very different and hard to guess for WSD devices.

I tested with `airscan-discover` when I was in the same subnet of my  printer devices

```
[devices]
  Brother MFC-T800W = http://172.****:80/WebServices/ScannerService, WSD
  FX DocuPrint M268 dw = http://172.****:80/WebServices/ScannerService, WSD
```

it seems that both `Brother MFC-T800W` and  `FX DocuPrint M268 dw` use **`/WebServices/ScannerService, WSD`** as the path part  

this can be a reference for someone else that can not run  `airscan-discover`  in the same broadcast LAN

Thanks for `alexpevzner`,again