alexpevzner/sane-airscan #344: Kyocera TASKalfa 3051ci issue

bernhardcygan, 2024-07-22:
The scanner is recognized in simple-scan, but scanning is not working. 

Attaching the support files in the hope that those shed some light on this. 

[simple-scan-Kyocera-TASKalfa-3051ci.log](https://github.com/user-attachments/files/16331556/simple-scan-Kyocera-TASKalfa-3051ci.log)
[simple-scan-zeroconf.log](https://github.com/user-attachments/files/16331557/simple-scan-zeroconf.log)

bernhardcygan, 2024-07-22:
Actually, after some further checking, scanning does work if I initiate it on the machine. This seems to be as designed.

alexpevzner, 2024-07-22:
Hi @bernhardcygan,

yes, this is how Kyoceras work. It is not very convenient, but nothing can be done with this. Newer Kyoceras also support eSCL, and in the eSCL mode they don't need scanning to be initiated on machine. I don't know why such a behaviour between otherwise such a similar protocols. Must other scanners don't require any buttons to be pressed on a front panel neither in the eSCL nor in the WSD mode.

closed 2024-07-22 by alexpevzner