alexpevzner/sane-airscan #356: Large scan jobs

Michael-EG, 2024-12-10:
I am testing with Ricoh MP 501 using adf, the issue is the scan job gets terminated, i think due to a timeout if scan job is too large (more than 50 pages) taking more than 3 or 4 minutes. Is there a parameter to increase timeout?

Michael-EG, 2024-12-11:
Thank you for your reply, i have sent you an email with a wireshark protocol report between PC and Ricoh MP 501 scanner.  Looking forward for your reply, Thank you for your assistance.

alexpevzner, 2024-12-11:
Please, in the `/etc/sane.d/airscan.conf` file uncomment the following lines:

```
[debug]
trace = ~/airscan/trace
```

Then rerun the test. It will automatically create the `~/airscan/trace` directory and leave very detailed logs (I call it proticol traces) there.

For now, `*.tar` files are not needed, I want to see `*.log` files only.

Michael-EG, 2024-12-11:
Thank you for your guidance on configuring the /etc/sane.d/airscan.conf file. I have followed your instructions to uncomment the necessary lines.

I greatly appreciate your expertise and support in helping to resolve this matter. Please let me know if there’s anything else I need to adjust or further actions required on my end.

Looking forward to your feedback.
[scanimage-RICOH-MP-601-[5838790CA68B].log](https://github.com/user-attachments/files/18093818/scanimage-RICOH-MP-601-.5838790CA68B.log)

alexpevzner, 2024-12-11:
Ho @Michael-EG,

thank for your respose!

I want to ask you to update config as follows (note added `hexdump` parameter) and to rerun the test:

```
[debug]
trace   = ~/airscan/trace
hexdump = true
```

Michael-EG, 2024-12-11:
Hi @alexpevzner,
Thank you so much for you assistance and patience. I attached an updated log file.
Looking forward to your feedback.
[scanimage-RICOH-MP-601-[5838790CA68B].log](https://github.com/user-attachments/files/18096495/scanimage-RICOH-MP-601-.5838790CA68B.log)

alexpevzner, 2024-12-11:
Hi @Michael-EG,

the situation is following. According to the log, scanner reports its capabilities etc etc etc and accepts scan job. Everything is OK at this point.

On request to download the scanned image, scanner nearly immediate returns response header (all eSCL communications happens via HTTP), then "thinks" about 3 minutes and after that unexpectedly drops the connection. Which `sane-airscan` interprets as an error.

Then `sane-airscan` cancels the failed job, and it also happens normally, and after that it reports the error.

So I have some questions:
* does scanning actually happens? Do you hear the scanning noise, does it consume paper from the feeder?
* does it scan from the flatbad?
* could there be a mechanical error, like paper jam or so on?

Michael-EG, 2024-12-13:
In my case, the scanner works fine and continues scanning a few pages even after the batch terminates. Here's what we've observed:

Scanning Activity: Yes, scanning does happen as expected. I hear the scanning noise, and the scanner does consume paper from the feeder without any apparent mechanical errors like paper jams.

Flatbed Scanning: Scanning from the flatbed works perfectly as well.

Job Termination Issue: The scanner sometimes continues the scanning job and throws an error at the end of the scanning job (SC899-00 a communication error) and other times the batch is terminated and the scanner continues for few pages and stops scanning but no paper jam. Instead, the batch is terminated by sane-airscan, which is unable to download the scanned file from the scanner. This happens after the scanner sends the response header but then delays for about 3 minutes before unexpectedly dropping the connection.

Looking forward to your insights.

alexpevzner, 2024-12-13:
May I ask you to show me the output of the following command:
```
airscan-discover
```

alexpevzner, 2024-12-14:
And another question.

If you insert only 1-2 pages into the ADF, does it work properly or not?

Michael-EG, 2024-12-15:
Here is the output for airscan-discover
`$ airscan-discover
[devices]
  RICOH MP 601 [5838790CA68B] = http://192.168.8.174:80/eSCL/, eSCL
`

When scanning few pages (maximum 50 pages) it works perfectly fine, the issue starts when the scan job i too large

alexpevzner, 2024-12-22:
@Michael-EG, ping...

Michael-EG, 2024-12-24:
sorry for late reply. I tried a fresh install for sane-airscan. However, same issue occurs i can not successfully scan large scan jobs. Working perfectly with small scan jobs (approximately 50 duplex pages or 100 one sided pages).

alexpevzner, 2024-12-27:
Hi @Michael-EG,

The issue occurs only when scanning large jobs, which seems to be related to the exhaustion of some resources (most likely memory) within the scanner. After some investigation, my plan was to limit the number of pages scanned from the ADF at once by using the `scan:NumberOfPages` parameter defined by the eSCL specification.

Unfortunately, your printer ignores this parameter (by the way, mine does too), so I  have to give up here. I apologize for that. 

The only recommendation I can make is to upgrade the printer firmware, unless you already have the latest version installed.

closed 2024-12-27 by alexpevzner