alexpevzner/sane-airscan #357: The Automatic Document Feeder (ADF) is not working properly on the EPSON WF-2930

Tony592, 2024-12-12:
I have a multifunction printer (Epson WF-2930). When I try to scan a multi-page document using the document feeder, only the first page is scanned. When debugging, it appears that after the first page, the feeder indicates 'out of document'.
For completeness, and to rule out hardware problem, I tried using EpsonScan2 sw and the Epson app, and the ADF works fine.
[scan.log](https://github.com/user-attachments/files/18110838/scan.log)

alexpevzner, 2024-12-12:
Hi @Tony592,

it doesn't look like `sane-airscan` log. Show me the output of the following command, please:
```
scanimage -L
```

Tony592, 2024-12-12:
Sure, here: 
device `airscan:e0:EPSON WF-2930 Series' is a eSCL EPSON WF-2930 Series ip=192.168.1.9

alexpevzner, 2024-12-12:
OK. Now I need the protocol trace.

In the `/etc/sane.d/airscan.conf` file, please, uncomment the following lines:
```
[debug]
trace = ~/airscan/trace
```
Then rerun the test. It will automatically create the `~/airscan/trace` directory and leave very detailed logs (I call it protocol traces) there.

For now, ***.tar** files are not needed, I want to see ***.log** files only.

Tony592, 2024-12-12:
[simple-scan-EPSON-WF-2930-Series.log](https://github.com/user-attachments/files/18112358/simple-scan-EPSON-WF-2930-Series.log)
[simple-scan-zeroconf.log](https://github.com/user-attachments/files/18112359/simple-scan-zeroconf.log)

alexpevzner, 2024-12-12:
OK. Are you able to build `sane-airscan` by yourself?

alexpevzner, 2024-12-15:
One more question.

If you scan with airscan from ADF, how many pages of paper does it physically consume?

Tony592, 2024-12-15:
All pages that are in the ADF

Tony592, 2024-12-15:
I don't know if It is relevant but when i scan, on the printer's display the string "scanning" appears only during the scanning of the first page while the ADF phisycally consumes all the others pages.

alexpevzner, 2024-12-16:
I.e., it consumes all pages, but creates an image only from the first page, correct?

And yet another question. If you will insert only 2 pages into the ADF, will it return both or the only first of them?

Tony592, 2024-12-16:
It Is Correct. 
Regarding the second questione: if I insert two pages in the ADF only the image of the first image is created.

Tony592, 2024-12-16:
Also using WSD protocl it doesn't woks

alexpevzner, 2024-12-17:
Hi @Tony592,

now I want to ask you  to fetch the latest GitHub version of the `sane-airscan` and test again.

The expected change: device will still scan only a single page, but will not consume all other pages from the ADF.

Installation is really simple. Just install gcc and all required libraries, enter the source directory and run `make` command.

Usually I don't recommend to perform `make install`, but instead recommend to locate where your distro install the `libsane-airscan.so.1` (typically it is `/usr/lib64/libsane-airscan.so.1` or `/usr/lib/libsane-airscan.so.1`, depending on distro) and replace it with the symbolic link to the full path of the `libsane-airscan.so.1` in your build directory.

Tony592, 2024-12-17:
Done, but also with the last update the ADF takes all pages

alexpevzner, 2024-12-17:
May I see the updated protocol traces? And from the WSD scan too, please.

Tony592, 2024-12-17:
Here:

[eSCL_simple-scan-EPSON-WF-2930-Series.log](https://github.com/user-attachments/files/18166261/eSCL_simple-scan-EPSON-WF-2930-Series.log)
[WSD_simple-scan-EPSON-WF-2930-Series.log](https://github.com/user-attachments/files/18166262/WSD_simple-scan-EPSON-WF-2930-Series.log)

alexpevzner, 2024-12-17:
OK. The next experiment, please.

In the file `airscan-wsd.c`, line 950 find the following line:
```
xml_wr_add_text(xml, "sca:ImagesToTransfer", "0");
```

Replace there `"0"` with `"1"`, rebuild and retest with WSD. Then send me the WSD log, please.

Tony592, 2024-12-17:
Here:
[simple-scan-EPSON-WF-2930-Series.log](https://github.com/user-attachments/files/18168725/simple-scan-EPSON-WF-2930-Series.log)

alexpevzner, 2024-12-17:
Still consumes all pages from the ADF?

Tony592, 2024-12-17:
Yes

alexpevzner, 2024-12-22:
Hi!

The next idea requires a noticeable amount of work in the `sane-airscan`, to to try it I wrote a small test in Go.

Please, do the following:
```
git clone https://github.com/alexpevzner/escl-test.git
cd escl-test
make
./escl-test http://192.168.1.4:443/eSCL
```

I assume your scanner's IP address is `192.168.1.4`, as in the log. And you will need a Golang compiler, basically any more-or-less recent version (1.18 or newer) will do.

Please let it run until completion - it may take seconds if not minutes, depending on your scanner. It will create the `escl-test.log`, please upload it here.

Tony592, 2024-12-22:
The ip is correct but when i launch the test I recive an error regarding a certificate not valid, i'll post the log
[escl-test.log](https://github.com/user-attachments/files/18222906/escl-test.log)

the printer is stuck on ‘scan’ for a while.

alexpevzner, 2024-12-22:
Fixed, please update and retest (if printer will not recover in a couple of minutes, you will need to reboot it)

Tony592, 2024-12-22:
Here the new log.
[escl-test.log](https://github.com/user-attachments/files/18222956/escl-test.log)

alexpevzner, 2024-12-22:
Please, update and retest.

Tony592, 2024-12-22:
Here:
[escl-test.log](https://github.com/user-attachments/files/18223051/escl-test.log)

alexpevzner, 2024-12-22:
Please, once again, update and retest.

Tony592, 2024-12-22:
here:
[escl-test.log](https://github.com/user-attachments/files/18223302/escl-test.log)

alexpevzner, 2024-12-24:
Hi,

I probably have no more idea, what can be done here. Looks like your scanner scans the entire ADF load, but after downloading the first page cancels remaining scan.

I've tried to limit ADF scan by a single page, but your scanner ignores the corresponding parameter. I also investigated the idea, that your scanner may consider TCP disconnect as a reason to cancel the scan job in progress (`sane-airscan` doesn't implement HTTP connection keep-alive, so every image loaded by establishing a new TCP connection), but experiments with the  `escl-test` program discovered, that connection is closed after each image download by the scanner itself.

So I have to give up here, sorry :(

closed 2024-12-24 by alexpevzner

Tony592, 2024-12-24:
I will use the Epson scan app, thanks for your work and patience.

alexpevzner, 2024-12-24:
One more thing, forgot to mention.

This is cleanly the firmware bug. It violates the eSCL specification. Have you tried to upgrade scanner firmware?

Tony592, 2024-12-24:
Yes, it has the last version, i tried to look the source code of the epson app but onestly I don't kow how to move.

alexpevzner, 2024-12-24:
Epson app uses its own proprietary protocol, while sane-airscan implements two generic protocols, eSCL and WS-Scan.

So looking to epson app sources will not help a lot to find workaround for eSCL/WSD.

I wonder if VueScan works well with your device and does it use eSCL or proprietary protocol (https://www.hamrick.com/)

alexpevzner, 2024-12-24:
Can you attach here the `.vuescan/vuescan.log` file if one exists?

Tony592, 2024-12-24:
Her
[vuescan.log](https://github.com/user-attachments/files/18239925/vuescan.log)

alexpevzner, 2024-12-24:
Looks like it uses Epson's proprietary protocol, not eSCL

Tony592, 2024-12-24:
Ok, thanks. We tried :)

Tony592, 2024-12-24:
With this device for now the proprietary driver is the only solution, the good thing is that at least it works.

alexpevzner, 2024-12-24:
OK. Have a good day! :)

maurerle, 2025-05-02:
Thanks for investigating this.
I also have an EPSON WF-2935 (so WF-2930 Series) which also does only scans the first page on ADF but consumes all pages in ADF.
I tried using their Firmware-Version:
`IP 180A`
after an update to version `IP 20P2` everything behaves the same.

Unfortunately, there are no firmware release notes.

maurerle, 2025-05-02:
I just tried using a MacBook using latest `macOS 15 Sequoia` trying to scan from the ADF there - as expected, the same behavior can be experienced.

I also tried to forward this issue to EPSON through the english CustHelp portal an the German Support portal, however I doubt it will find hearing there, but eventually a new EPSON firmware version will include a fix..