alexpevzner/sane-airscan #358: ADF Source Option Missing for Epson WorkForce 845

jonmchan, 2024-12-22:
The ADF option is missing for the Epson WorkForce 845 printer.

```
scanimage -L
device `airscan:w0:EPSON WorkForce 845' is a WSD EPSON WorkForce 845 ip=192.168.62.10

scanimage --help -d "airscan:w0:EPSON WorkForce 845"

..
    --source Flatbed [Flatbed]
        Selects the scan source (such as a document-feeder).
..
```

The epson2 driver shows availability for the ADF:

```
scanimage -d "epson2:net:192.168.62.10" -L
device `epson2:net:printer.cf.home' is a Epson PID 0892 flatbed scanner

scanimage --help -d "epson2:net:192.168.62.10"
    --source Flatbed|Automatic Document Feeder [Flatbed]
        Selects the scan source (such as a document-feeder).
```

Any chance that airscan can enable ADF support for this printer or is this a limitation of what is exposed in the WSD interface? Thank you.

alexpevzner, 2024-12-22:
Hi @jonmchan,

I need to see the protocol trace.

To obtain it, in the `/etc/sane.d/airscan.conf` file, please, uncomment the following lines:
```
[debug]
trace = ~/airscan/trace
```
Then rerun the test. It will automatically create the `~/airscan/trace` directory and leave very detailed logs (I call it protocol traces) there.

For now, ***.tar** files are not needed, I want to see ***.log** files only.

jonmchan, 2024-12-22:
Thank you for looking into this.

[scanimage-EPSON-WorkForce-845.log](https://github.com/user-attachments/files/18222542/scanimage-EPSON-WorkForce-845.log)

jonmchan, 2024-12-22:
Yes, I loaded the paper when testing. I did notice that as well that there was no mention of ADF source in the response. 

Here's the logs when I take the paper out and I put the paper in - it doesn't look promising.

[scanimage-EPSON-WorkForce-845.no-adf-load.log](https://github.com/user-attachments/files/18222896/scanimage-EPSON-WorkForce-845.no-adf-load.log)
[scanimage-EPSON-WorkForce-845.paper-loaded-adf.log](https://github.com/user-attachments/files/18222897/scanimage-EPSON-WorkForce-845.paper-loaded-adf.log)

alexpevzner, 2024-12-22:
My Kyocera doesn't sense paper load into the ADF when it is sleeping. So may I suggest you to wakeup your printer (say, by printing or scanning something), then try to load the ADF?

But in general, yes, there is no too much hope to get this device work properly with ADF via WSD... :(

jonmchan, 2024-12-23:
@alexpevzner I appreciate your time and efforts in debugging this with me. I think the Epson Workforce 845 just did not implement WSD too well. Thank you.

For anyone who googles this in the future, I did get this successfully working with the epkowa driver, but it wasn't available from debian repo, you had to get it directly from epson. I used the plugin from the iscan-bundle.

closed 2024-12-23 by jonmchan