alexpevzner/sane-airscan #362: Help! Brother DS-940DW - Document feeder out of documents

ProvoK, 2025-01-19:
Hi there, 
coming from https://github.com/sbs20/scanservjs which - if I understood correctly - is just a frontend for `scanimage`, which is developed here. Please keep me honest on this

I'm getting some errors trying to scan using a Brother DS-940DW (installed latest firmwares), I'm always getting the "Document feeder out of documents" error.

I've tried:
1) installing the latest firmware on the scanner
2) installing the latest driver on the system (deb pkg)
3) using different outputs formats
4) Different SANE frontend: NAPS2 works (screenshot)

[image]

One thing I noticed is the different naming of the paper source 

[image]

```
root@3eb49463102a:/usr/lib/scanservjs# scanimage --device airscan:e1:DS-940DW -A
Output format is not set, using pnm as a default.

All options specific to device `airscan:e1:DS-940DW':
  Standard:
  [...]
    --source ADF|ADF Duplex [ADF]
        Selects the scan source (such as a document-feeder).
  [...]
```

Any pointer/help on this? 🙏 

Below, I'm pasting the debug logs of the scanimage command issued.
```
root@ac5308fa0c8f:/usr/lib/scanservjs# scanimage --device airscan:e1:DS-940DW --format pdf -o my.pdf
API: sane_init(): called
loading configuration file /etc/sane.d/airscan.conf
MDNS: devices discovery disabled
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
WSDD: devices discovery disabled
zeroconf: ZEROCONF_WSD: initial scan finished
API: sane_open("e1:DS-940DW"): called
zeroconf: device_list wait: requested
zeroconf: device_list wait: OK
DS-940DW: device created
DS-940DW: DEVICE_STM_OPENED->DEVICE_STM_PROBING
DS-940DW: using protocol "eSCL"
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerCapabilities
DS-940DW: HTTP using timeout: 5000 ms
DS-940DW: HTTP resolving 192.168.1.68 80
DS-940DW: HTTP trying 192.168.1.68:80
DS-940DW: HTTP fdpoll: {--} -> {-w}
DS-940DW: HTTP 81 bytes sent
DS-940DW: HTTP done request sending
DS-940DW: HTTP fdpoll: {-w} -> {rw}
DS-940DW: HTTP fdpoll: {rw} -> {r-}
DS-940DW: HTTP 2920 bytes received
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerCapabilities: got response headers (200)
DS-940DW: HTTP 2624 bytes received
DS-940DW: HTTP 4147 bytes received
DS-940DW: HTTP done response reception
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerCapabilities: 200 OK
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerCapabilities: OK
DS-940DW: new decoder: jpeg
DS-940DW: DEVICE_STM_PROBING->DEVICE_STM_IDLE
DS-940DW: API: sane_open("e1:DS-940DW"): Success
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: get (0): 19
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: get (0): 19
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: get br-x: 215.9
DS-940DW: API: get tl-x: 0
DS-940DW: API: get br-y: 3098.8
DS-940DW: API: get tl-y: 0
DS-940DW: API: get tl-x: 0
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: set br-x: 215.9
DS-940DW: API: get tl-y: 0
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: set br-y: 3098.8
DS-940DW: API: device_get_option_descriptor(): called
DS-940DW: API: device_get_option_descriptor(): done
DS-940DW: API: get resolution: 300
DS-940DW: API: sane_start(): called
DS-940DW: device_start_wait: waiting
DS-940DW: DEVICE_STM_IDLE->DEVICE_STM_SCANNING
DS-940DW: PROTO_OP_PRECHECK: submitting: attempt=0
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerStatus
DS-940DW: HTTP using timeout: 5000 ms
DS-940DW: HTTP resolving 192.168.1.68 80
DS-940DW: HTTP trying 192.168.1.68:80
DS-940DW: HTTP fdpoll: {--} -> {-w}
DS-940DW: HTTP 75 bytes sent
DS-940DW: HTTP done request sending
DS-940DW: HTTP fdpoll: {-w} -> {rw}
DS-940DW: HTTP fdpoll: {rw} -> {r-}
DS-940DW: HTTP 816 bytes received
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerStatus: got response headers (200)
DS-940DW: HTTP done response reception
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerStatus: 200 OK
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerStatus: OK
DS-940DW: PROTO_OP_PRECHECK: decoding
DS-940DW: PROTO_OP_PRECHECK: device status: Success
DS-940DW: PROTO_OP_PRECHECK: ADF status: Document feeder out of documents
DS-940DW: PROTO_OP_PRECHECK: decoded: status="Success" next=PROTO_OP_SCAN delay=0
DS-940DW: PROTO_OP_SCAN: submitting: attempt=0
DS-940DW: HTTP POST http://192.168.1.68/eSCL/ScanJobs
DS-940DW: HTTP using timeout: 30000 ms
DS-940DW: HTTP resolving 192.168.1.68 80
DS-940DW: HTTP trying 192.168.1.68:80
DS-940DW: HTTP fdpoll: {--} -> {-w}
DS-940DW: HTTP 891 bytes sent
DS-940DW: HTTP done request sending
DS-940DW: HTTP fdpoll: {-w} -> {rw}
DS-940DW: HTTP fdpoll: {rw} -> {r-}
DS-940DW: HTTP 112 bytes received
DS-940DW: HTTP POST http://192.168.1.68/eSCL/ScanJobs: got response headers (404)
DS-940DW: HTTP done response reception
DS-940DW: HTTP POST http://192.168.1.68/eSCL/ScanJobs: 404 Not Found
DS-940DW: HTTP POST http://192.168.1.68/eSCL/ScanJobs: Not Found
DS-940DW: PROTO_OP_SCAN: decoding
DS-940DW: PROTO_OP_SCAN: decoded: status="Success" next=PROTO_OP_CHECK delay=0
DS-940DW: ScanJobs request: unexpected HTTP status 404
DS-940DW: PROTO_OP_CHECK: submitting: attempt=0
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerStatus
DS-940DW: HTTP using timeout: 5000 ms
DS-940DW: HTTP resolving 192.168.1.68 80
DS-940DW: HTTP trying 192.168.1.68:80
DS-940DW: HTTP fdpoll: {--} -> {-w}
DS-940DW: HTTP 75 bytes sent
DS-940DW: HTTP done request sending
DS-940DW: HTTP fdpoll: {-w} -> {rw}
DS-940DW: HTTP fdpoll: {rw} -> {r-}
DS-940DW: HTTP 816 bytes received
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerStatus: got response headers (200)
DS-940DW: HTTP done response reception
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerStatus: 200 OK
DS-940DW: HTTP GET http://192.168.1.68/eSCL/ScannerStatus: OK
DS-940DW: PROTO_OP_CHECK: decoding
DS-940DW: PROTO_OP_CHECK: device status: Success
DS-940DW: PROTO_OP_CHECK: ADF status: Document feeder out of documents
DS-940DW: PROTO_OP_CHECK: job status: Document feeder out of documents
DS-940DW: PROTO_OP_CHECK: decoded: status="Document feeder out of documents" next=PROTO_OP_FINISH delay=0
DS-940DW: JOB status=Document feeder out of documents
DS-940DW: DEVICE_STM_SCANNING->DEVICE_STM_DONE
DS-940DW: device_start_wait: Document feeder out of documents
DS-940DW: DEVICE_STM_DONE->DEVICE_STM_IDLE
DS-940DW: API: sane_start(): Document feeder out of documents
scanimage: sane_start: Document feeder out of documents
DS-940DW: API: sane_close(): called
DS-940DW: DEVICE_STM_IDLE->DEVICE_STM_CLOSED
DS-940DW: removed from device table
DS-940DW: closed protocol "eSCL"
DS-940DW: closed decoder: jpeg
DS-940DW: device destroyed
DS-940DW: API: sane_close(): done
API: sane_exit(): called
API: sane_exit(): OK
```

alexpevzner, 2025-01-19:
Hi @ProvoK,

I need the protocol trace. To obtain it, you should uncomment in the `/etc/sane.d/airscan.conf` file the following lines:
```
[debug]
trace   = ~/airscan/trace
```

The `~/airscan/trace` directory will be created automatically, and the protocol trace files will be written there. For now, I only need the `*.log` files, `*.tar` files are not needed.

ProvoK, 2025-01-20:
Hey @alexpevzner 

Sure, here they are!

[scanimage-DS-940DW.log](https://github.com/user-attachments/files/18472247/scanimage-DS-940DW.log)
[scanimage-zeroconf.log](https://github.com/user-attachments/files/18472248/scanimage-zeroconf.log)

alexpevzner, 2025-01-20:
It looks like you actually work with `HP ScanJet Pro 3500 f1`, not with `Brother DS-940DW`...

May I see output of the `airscan-discover` command?

ProvoK, 2025-01-20:
yes, I've also the HP printer/scanner, that works fine though.

The [scanimage-DS-940DW.log](https://github.com/user-attachments/files/18472247/scanimage-DS-940DW.log) should contain the appropriate logs. Keep me honest on this 🙏 

> May I see output of the airscan-discover command?

Please note that I had to disable discovery and do manual setup of devices: it was unable to find any device on my local network

```
root@bcb29b53f3fb:/usr/lib/scanservjs# /usr/bin/airscan-discover -d -t 
MDNS: avahi_client_new failed: Daemon not running
[devices]
```
This is my airscan config
```
root@bcb29b53f3fb:/usr/lib/scanservjs# cat /etc/sane.d/airscan.conf 
[devices]
HP = http://192.168.1.38/eSCL, eSCL
DS-940DW = http://192.168.1.68/eSCL, eSCL

[options]
discovery = disable

[debug]
trace = /tmp/airscan
```

ProvoK, 2025-01-20:
I misread, I though you referred to the other HP machine.

[image]

This is fun: probably they whitelabeled a HP product as Brother? 😂 Or copy pasted something?
The IP is definitely the one from the Brother portable scanner

alexpevzner, 2025-01-20:
According to the log you've attached, you actually work with HP:

```
<?xml version="1.0" encoding="UTF-8"?>
<scan:ScannerCapabilities xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03" xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm" xmlns:dest="http://schemas.hp.com/imaging/destination/2011/06/06" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://schemas.hp.com/imaging/escl/2011/05/03 ../../schemas/eSCL.xsd">
        <pwg:Version>2.63</pwg:Version>
        <pwg:MakeAndModel>HP ScanJet Pro 3500 f1</pwg:MakeAndModel>
        <pwg:SerialNumber>SN24</pwg:SerialNumber>
        . . . .
```

alexpevzner, 2025-01-20:
> This is fun: probably they whitelabeled a HP product as Brother?

This is very interesting. I've never seen anything like this. Vendors really sometimes relabel somebody's else product as their own (most mass marked Xerox and some HPs are actually Samsungs), but these two models look so different on  a picture...

I still want to be sure that we really work with device we think we work with.

airscan-discover complains that Avahi daemon is not running. Can you start Avahi and rerun  airscan-discover?

alexpevzner, 2025-01-20:
> I'll try: I'm inside a docker image so it might be a bit more complex to make it work

What is your host operating system?

ProvoK, 2025-01-20:
MacOS 14.4 (Intel)

alexpevzner, 2025-01-20:
Can you try the following commands on MacOS host?

```
dns-sd -B _uscan._tcp
dns-sd -B _uscans._tcp
```

ProvoK, 2025-01-20:
```
$ dns-sd -B _uscan._tcp
Browsing for _uscan._tcp
DATE: ---Mon 20 Jan 2025---
12:03:17.170  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
12:03:17.171  Add        2   7 local.               _uscan._tcp.         HPB0227A4E0347 (HP Laser MFP 131 133 135-138)
12:04:07.538  Add        2   7 local.               _uscan._tcp.         Brother DS-940DW [5CF370C2B191]
```

```
$ dns-sd -B _uscans._tcp
Browsing for _uscans._tcp
DATE: ---Mon 20 Jan 2025---
12:04:27.428  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
12:04:27.429  Add        3   7 local.               _uscans._tcp.        HPB0227A4E0347 (HP Laser MFP 131 133 135-138)
12:04:27.429  Add        2   7 local.               _uscans._tcp.        Brother DS-940DW [5CF370C2B191]
```

ProvoK, 2025-01-20:
I've also tried to redo the same `scanimage` after shutting down the HP printer, just in case. Same result

alexpevzner, 2025-01-20:
And these two, please:
```
dns-sd -L "Brother DS-940DW [5CF370C2B191]" _ucans._tcp
dns-sd -L "Brother DS-940DW [5CF370C2B191]" _uscans._tcp
```
Plaase notice the difference, "_uscan" vs "_uscans"

ProvoK, 2025-01-20:
ucans seems not to find it
```
$ dns-sd -L "Brother DS-940DW [5CF370C2B191]" _ucans._tcp
Lookup Brother DS-940DW [5CF370C2B191]._ucans._tcp.local
DATE: ---Mon 20 Jan 2025---
13:37:08.403  ...STARTING...                  
```

while uscans does
```
$ dns-sd -L "Brother DS-940DW [5CF370C2B191]" _uscans._tcp

Lookup Brother DS-940DW [5CF370C2B191]._uscans._tcp.local
DATE: ---Mon 20 Jan 2025---
13:37:39.848  ...STARTING...
13:37:39.985  Brother\032DS-940DW\032[5CF370C2B191]._uscans._tcp.local. can be reached at BR5CF370C2B191.local.:443 (interface 7)
 txtvers=1 vers=2.63 adminurl=http://BR5CF370C2B191.local./ representation=none rs=eSCL ty=Brother\ DS-940DW note= pdl=application/octet-string,image/jpeg UUID=437B5664-9208-4a8e-B85D-DDB189FCD934 is=adf duplex=T cs=color,grayscale,binary

```

alexpevzner, 2025-01-20:
```
dns-sd -G v4 BR5CF370C2B191.local.
```

ProvoK, 2025-01-20:
```
$ dns-sd -G v4 BR5CF370C2B191.local.

DATE: ---Mon 20 Jan 2025---
14:42:47.565  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname                               Address                                      TTL
14:42:47.566  Add  40000002       7  BR5CF370C2B191.local.                  192.168.1.68                                 120
```

alexpevzner, 2025-01-20:
OK.

Can you try in the `airscan.conf` `https://192.168.1.68/eSCL` instead of `http://192.168.1.68/eSCL`?

ProvoK, 2025-01-20:
That was it, I'm able to scan pages now - thanks a lot for the support!

Fun fact: I did actually try HTTPS before creating this issue, by adding another device (`[...]-https` in the airscan.conf).
But I got confused by this error
```
scanimage: open of device airscan:e1:DS-940DW-https failed: Invalid argument
```
I simply added `-https` to the device name when running `scanimage` (as I did in the config), but given the error I though it was incompatible with https for some reason.
Instead the issue was that it had to ba `e2` instead of `e1` 🤦 

Thanks again, yet another case of pebcak

closed 2025-01-20 by ProvoK

alexpevzner, 2025-01-24:
Could you please show me a protocol trace after successful scan, just for my collection?

ProvoK, 2025-01-28:
Sure, here it is!

[scanimage-DS-940DW-https.log](https://github.com/user-attachments/files/18579764/scanimage-DS-940DW-https.log)