alexpevzner/sane-airscan #375: Limited scan resolution

mb720, 2025-08-30:
Hi and thanks a lot for sane-airscan!

On Arch Linux using brscan4 I can scan with a resolution of up to 9600 DPI on a Brother DCP-L3550CDW.

With `scanimage -d 'brother4:net1;dev0' --all-options` I can see the supported scan resolutions:

    --resolution 100|150|200|300|400|600|1200|2400|4800|9600dpi [200]

On another laptop, also running Arch Linux, I'm using sane-airscan instead of brscan4 and `scanimage -d "airscan:e1:Brother DCP-L3550CDW series" --all-options` returns lower resolutions:

    --resolution 100|200|300|600dpi [300]

Indeed, while simple-scan offers scanning with 1200 DPI, the scanned images only have a resolution of 600 DPI.

Is there a way to configure sane-airscan to allow scanning with a higher resolution?

Thanks in advance and let me know if I can provide any further information to troubleshoot this.

alexpevzner, 2025-08-30:
Hi!

`sane-airscan` obtains list of supported resolutions from the device and exposes it to the client software as is. It doesn't imply its own limitations.

This is not uncommon for the scanners to expose more options via the proprietary interface rather that via standards-based protocols, like eSCL and WSD. Surprisingly,  it happens especially often with the inexpensive devices, like Brother.

But most likely, the actual optical resolution of your scanner is 600 DPI, while the fantastic 9600 DPI, offered by the proprietary driver is just the software interpolation.

Basically, you can get the same effect if you scan with the `sane-airscan` at the maximum possible resolution and then resample the resulting image into the higher resolution. But it will not add image details, it will only add brute amount of pixels.

The fact that `simple-scan` offers 1200 DPI is the bug in the `simple-scan`. The list of resolutions reported by the `sane-airscan` is the same, regardless of the client software being used.

mb720, 2025-08-31:
Thanks for your quick reply!

>But most likely, the actual optical resolution of your scanner is 600 DPI

Interesting! After doing some tests with the proprietary driver from brscan4, this seems plausible. I've used simple-scan to scan a photo, setting the "Compression" slider in the options to "Full detail". I selected 600 DPI, 1200 DPI and 2400 DPI in the GUI. Higher resolutions cause significantly longer scan times, suggesting to me that the scanner captures more details and this produces images with a higher DPI.

But while there seems indeed more details between 600 DPI and 1200 DPI, there are no more visual details between 1200 DPI and 2400 DPI.

Here are screenshots from the scanned image, viewed and zoomed in with feh:

600 DPI:

[image]

1200 DPI:

[image]

2400 DPI:

[image]

Given that there's additional detail between 600 DPI and 1200 DPI but not between 1200 DPI and 2400 DPI, maybe the actual maximum scan resolution is 1200?