alexpevzner/sane-airscan #378: eSCL on Xerox(R) B225 MFP / B235 works but not WSD

ruikruik, 2025-09-28:
Hi,

I have a partial success with The Xerox(R) B225 MFP or B235 MFP  using eSLC which works fine so far. The WSD has some issue(s). I tried with latest version of git (replaced the original one /usr/lib/x86_64-linux-gnu/sane/libsane-airscan.so.1) but it behaves same.

[devices]
  Xerox(R) B225 MFP = http://192.168.42.84:80/eSCL/, eSCL
  Xerox(R) B225 MFP = https://192.168.42.84:443/eSCL/, eSCL
  Xerox(R) B225 MFP = http://[XXXX:XXXX:5007:4500::7b4]:65004/Scanner1/WebServices, WSD
  Xerox(R) B225 MFP = http://192.168.42.84:65004/Scanner1/WebServices, WSD

The scanning only works in eSCL mode, the WSD mode causes scan utility to print an error and scanning never completes (printer reports "remote scan in progress". After this happens, the WSD disappears from "simple-scan"  choose scanner dialog, but the airscan-discover still reports that.

I can attach more of debug output, lets start with the simple case, where I choose WSD version of scanner which fails to scan (it prints error message in simple scan, but printer reports remote scanning is in progress) The .tar is empty.

[simple-scan-zeroconf.log](https://github.com/user-attachments/files/22583865/simple-scan-zeroconf.log)

I tried to look around the source for some Xerox quirks but I could not find any for the WSD. I will be happy to debug that if you can provide some hints what to try next (I don't own any other WSD device co I don't have some good log).

Thanks,
Rudolf

alexpevzner, 2025-09-28:
Hi!

To investigate this issue, I need another log file. The one that contains the printer name instead of the `zeroconf`. It should be automatically created after the scan attempt in the same directory where you've fould the `simple-scan-zeroconf.log` file.

ruikruik, 2025-09-29:
Hi thanks for the quick response and also for the sane-airscan :)

[debug]
trace   = ~/airscan/trace3-wsd

The trace directory has only two files:

ls -1
simple-scan-zeroconf.log
simple-scan-zeroconf.tar
The tar file is either zero size (directly after fail), or contains 1024 bytes of binary zeros (no bits are set) after I close the application.

The simple scan fails immediately when I hit the scan button and complaining that it cannot start scanning.

Thanks,
Rudolf

alexpevzner, 2025-09-30:
This is very strange. Can you enable console logging:
```
[debug]
enable = true
```

and send me console output?